pikvm/pikvm #614: mouse/keyboard emulation not working with OSX

cipy, 2022-01-18:
PiKVM doesn't seem to be able to control a MacBookPro (Intel or M1)

the video over HDMI is fine, but there's no interaction with the OSX

**To Reproduce**
Steps to reproduce the behavior:
1. use a MacBookPro Intel or M1
2. connect it to a working PiKVM HAT v3
3. try to interact using the mouse/keyboard
4. oops/fail

**Expected behavior**
Should work fine.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: macOS 12.0 Monterey or macOS 11.0 Big Sur
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version RPi 4B
  - PiKVM platform v3
  - Video capture type CSI bridge
  - latest `pacman -Syu`
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

Thanks,

arch1mede, 2022-01-18:
Hi, can you join discord so we can troubleshoot this, it seems like there is maybe an issue with the v3 usbc bridge or cable. I would rather not do troubleshooting in this space.

mdevaev, 2022-01-21:
@cipy Sup? We want to help you :)

cipy, 2022-01-21:
I'll join your Discord soon, thanks ☺️

mdevaev, 2022-01-25:
@cipy boop

mdevaev, 2022-01-26:
The problem is probably related to the fact that you have not installed a USB-C bridge.If you have any questions, please contact our Discord support. I'm closing it now because it's not a bug and we know that everything works with OSX.

closed 2022-01-26 by mdevaev