pikvm/pikvm #617: Incorrect 5Ghz wifi channels for EU

raqua, 2022-01-22:
**Describe the bug**
5Ghz wifi channels are not correct and I can not connect to my wifi, because it is on different channel. 
I am in Slovakia and I have set  "SK" in /etc/conf.d/wireless-regdom. I also tried to set country=SK in /etc/wpa_supplicant/wpa_supplicant-wlan0.conf.

```
[root@pikvm ~]# iw reg get 
global
country SK: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```

I am getting these channels:
```
[root@pikvm ~]# iwlist channel
lo        no frequency information.

eth0      no frequency information.

wlan0     22 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 149 : 5.745 GHz
          Channel 153 : 5.765 GHz
          Channel 157 : 5.785 GHz
          Channel 161 : 5.805 GHz
          Channel 165 : 5.825 GHz
          Current Frequency:2.452 GHz (Channel 9)

```

According to https://en.wikipedia.org/wiki/List_of_WLAN_channels#5_GHz_(802.11a/h/j/n/ac/ax), there should be much more channels for EU,  my wifi is for example on ch.128. This works fine in many other devices and is legal in EU.

I have one more RPi4 on with Raspbian and channels there look like this:
```
root@retropie:/# iwlist channel
eth0      no frequency information.

wlan0     32 channels in total; available frequencies :
          Channel 01 : 2.412 GHz
          Channel 02 : 2.417 GHz
          Channel 03 : 2.422 GHz
          Channel 04 : 2.427 GHz
          Channel 05 : 2.432 GHz
          Channel 06 : 2.437 GHz
          Channel 07 : 2.442 GHz
          Channel 08 : 2.447 GHz
          Channel 09 : 2.452 GHz
          Channel 10 : 2.457 GHz
          Channel 11 : 2.462 GHz
          Channel 12 : 2.467 GHz
          Channel 13 : 2.472 GHz
          Channel 36 : 5.18 GHz
          Channel 40 : 5.2 GHz
          Channel 44 : 5.22 GHz
          Channel 48 : 5.24 GHz
          Channel 52 : 5.26 GHz
          Channel 56 : 5.28 GHz
          Channel 60 : 5.3 GHz
          Channel 64 : 5.32 GHz
          Channel 100 : 5.5 GHz
          Channel 104 : 5.52 GHz
          Channel 108 : 5.54 GHz
          Channel 112 : 5.56 GHz
          Channel 116 : 5.58 GHz
          Channel 120 : 5.6 GHz
          Channel 124 : 5.62 GHz
          Channel 128 : 5.64 GHz
          Channel 132 : 5.66 GHz
          Channel 136 : 5.68 GHz
          Channel 140 : 5.7 GHz
          Current Frequency:2.452 GHz (Channel 9)
```

**To Reproduce**
Steps to reproduce the behavior, like:
1. Setup WIRELESS_REGDOM="SK"
2. Go to terminal and run 'iwlist channel'
3. See incorrect channels supported

**Expected behavior**
Correct channels should be available.

**Desktop (please complete the following information):**
Not applicable here

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform : v3
  - Video capture type : CSI bridge ??
  - KVMD version: 3.48
  - uStreamer version: 4.9
  - Linux kernel: pikvm 5.10.52-13-ARCH

**Additional context**

arch1mede, 2022-01-23:
Just adding my 2 cents, this isn't a PiKVM issue, this is a arch arm linux issue and should be directed towards that group.

PiKVM is software that's riding on top of an existing OS, this being arch arm linux.

arch1mede, 2022-03-04:
Was there any more context to this that you could provide? Like what channel specifically you are looking for? Or why you cannot connect? Or have you been able to connect since last posting? Or are you just reporting the channel differences?

If you look at both those outputs you can see that both arch arm and rasbian are reporting channels it can and cant see, differently. EX: Rasbian reports seeing channel 100, arch arm does not, arch arm is reports seeing channel 165 but rasbian doesn't. I think that's some sort of cache as it displays regardless on the state of wlan0.

raqua, 2022-03-04:
My wifi is running at the channel 128 and I can connect on the same RPi by just swapping the SD card with Raspbian. 
I was not able to connect and I am using 2.4GHz wifi. 

"Rasbian reports seeing channel 100, arch arm does not, arch arm is reports seeing channel 165 but rasbian doesn't. I think that's some sort of cache as it displays regardless on the state of wlan0."
-> I believe this is some sort of regional setting, maybe coming from wireless regdom, but not correctly applied in Arch? Raspbian is correct here.

mdevaev, 2022-03-21:
Please try to install the latest OS image (or update an existent OS) and configure your regdomain on it. I updated kernel and it may be fixed already.

raqua, 2022-03-21:
I can try that, will report back, but I am rather skeptical that this is kernel issue. I would say it is wifi stack config.

raqua, 2022-03-28:
Hi, sorry for late reply, I was busy with other stuff. I have just updated the system and there is no change.

mdevaev, 2022-03-28:
To be honest I have no idea how to fix it.

Maybe @xornet-sl could tell something.

arch1mede, 2022-03-31:
So rather than leaving you hanging on this issue:

unfortunately we doesn't know how to fix it because this problem is laying in the base of the Arch Linux, so we recommend you to ask about this problem in the Arch Linux ARM forum, they will help you better then we do

raqua, 2022-03-31:
OK. It is still usable with 2.4GHz, so not huge issue. 
I will report here if I figure it out.

closed 2022-03-31 by mdevaev