pikvm/pikvm #628: KVM Switch Video Problems

mgeorgy, 2022-01-28:
Hello.  I know that not every KVM Switch is supported by the PiKVM.  I'm hoping there is a solution to problem I'm experiencing with my setup.  When switching between video sources I get a distorted image.  I currently have to reboot the PiKVM for the stream to clear up.  Is there a way to completely reset the CSI or otherwise simulate disconnecting and reconnecting the HDMI cable?

[image]

arch1mede, 2022-02-01:
Try playing with the [edid](https://docs.pikvm.org/edid/) otherwise our support is limited and we would say that whatever kvm you are using is not supported or the cable is bad.

mdevaev, 2022-02-05:
V3 HAT has the hardware ability to reset the HDMI chip, but I'm not sure that this will help. Let there be a feature, at some point we will implement it.

sebtech33, 2022-06-10:
Would you mind just adding the name and model of the kvm you used so other could be aware to?

cgm999, 2022-12-01:
Hello,

Please share how we can restart the tc358743 Toshiba chip on V3 hat.. from time to time it gets confused and it would be nice to do that (is that done when at Reset stream?)

mdevaev, 2022-12-02:
Right now the best way is rebooting entire pikvm.

aidv, 2023-07-07:
> Right now the best way is rebooting entire pikvm. Пт, 2 дек. 2022 г. в 01:26, cgm999 ***@***.***>:

Disconnecting then re-connecting the HDMI cable seems to be sufficient.