pikvm/pikvm #632: routing via the PiKVM is not working

cipy, 2022-02-02:
The how-to at https://docs.pikvm.org/usb_ethernet/#routing-via-pikvm is a bit wrong as the firewall (in the INPUT chain) drops everything 

```
iptables -L -n
Chain INPUT (policy ACCEPT)
target     prot opt source               destination        
ACCEPT     icmp --  0.0.0.0/0            0.0.0.0/0          
ACCEPT     udp  --  0.0.0.0/0            0.0.0.0/0            udp dpt:67
DROP       all  --  0.0.0.0/0            0.0.0.0/0          

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination        
ACCEPT     all  --  0.0.0.0/0            0.0.0.0/0          

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

arch1mede, 2022-02-03:
Related #282

mdevaev, 2025-11-16:
It was fixed in #957 a long time ago.

closed 2025-11-16 by mdevaev