pikvm/pikvm #633: mouse & keyboard not working on the iPad Pro

cipy, 2022-02-02:
PiKVM v3 HAT connected using [Apple's official USB-C dongle for HDMI+USB](https://www.apple.com/shop/product/MUF82AM/A/usb-c-digital-av-multiport-adapter) to an iPad Pro 2020 with iPadOS 15.3 doesn't seem to be able to transmit mouse & keyboard interactions to it.

P.S. the video in the browser has vertical black bars on the sides

arch1mede, 2022-02-03:
[Related](https://github.com/pikvm/pikvm/issues/614#issuecomment-1027919493)

cipy, 2022-02-03:
a bit related but not much, as this time the PiKVM has the USB-C bridge, but maybe iOS 15.3 and/or the dongle is not happy with these software emulated devices ❓ @arch1mede

mdevaev, 2022-02-05:
@cipy try the single relative mouse mode: https://docs.pikvm.org/mouse

arch1mede, 2022-02-05:
I use [this](https://www.amazon.com/gp/product/B0819KTP6M) and it works.

arch1mede, 2022-02-05:
> 

Your right, at the time though you didn't have the usbc bridge working when you posted this. I have removed my comment as a result.

mdevaev, 2022-02-09:
Apparently the problem has been solved

closed 2022-02-09 by mdevaev