pikvm/pikvm #634: zerotier support

ahmaddxb, 2022-02-03:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

I have installed zerotier on to the pikvm and joined it to my network but it doesnt connect, always shows offline and doesnt appear in myzerotier central account to authorise.

Any Ideas why?

arch1mede, 2022-02-03:
Just my 2 cents, tailscale document was only added due to how easy it was to deploy and use so was added as a courtesy, zerotier needs more steps and is not as easy, IMO this should not be added to the current document, this is arch arm linux so its best you ask in whatever support structure they currently have.

mdevaev, 2022-02-05:
Unfortunately, we have no experience in setting up zerotier. You should pay either to zerotier support, or to arch linux (pikvm is actually arch linux arm). Zerotier's technical support is beyond our competence.

closed 2022-02-05 by mdevaev

shinji257, 2022-08-09:
This is an old issue but I figured this is the best place to drop this.  Keep in mind that this setup is still _not supported_ as per the earlier message.  I'm not actually offering my own support for this but thought to throw some pointers for others.  Anyways I managed to get it to work.

Main points here.  

* During setup and any reconfiguration, the system has to be in rw mode due to needing to write into /var/lib/zerotier-one folder to store said configuration. 
* After installing you have to enable and start the associated service (it is disabled by default with the arch linux package)
* Service needs a working /dev/net/tun device present on the system.  If it isn't there after install, then reboot the system to create it.  Keep the system in rw mode for this purpose specifically so that it can be created.

Once you have the service running and the tun device created you should be able to join networks and get it authorized while in rw mode.  After all configuration is done and you have verified it works then you can move to ro mode.  Most likely you can ignore warnings from it regarding the inability to write to associated files... I think.  If the connection with zerotier breaks at some point, then you will have to switch the system back to rw mode so that zerotier can refresh its mesh nodes.

A useful and related wikidoc for zerotier on Arch Linux
https://wiki.archlinux.org/title/ZeroTier

ahmaddxb, 2022-08-09:
This is no longer an issue any more, It has been working for a while now, maybe after one of the previous system updates.

kaaass, 2023-06-01:
Thanks for the experience from @shinji257! I was able to establish a functioning Zerotier service. However, there's still a problem when it comes to getting Zerotier to launch automatically at system boot time.

Zerotier needs to write to the `/var/lib/zerotier-one/authtoken.secret` file upon start-up, which necessitates temporarily switching the system to rw mode. Here's a workaround I've devised to tackle this:

1. Run `rw` to switch to the rw mode. This is necessary as it allows us to edit the service unit file.
2. Start editing the Zerotier service unit with `systemctl edit zerotier-one`.
3. Paste the following code into the editor. This overrides the existing `ExecStart` command in the service definition. The new `ExecStart` command ensures that we switch to rw mode, start the Zerotier service, wait for 30 seconds (enough time to allow Zerotier to write its required files, at least in my experience), and then switch back to ro mode:

```
[Service]
ExecStart=
ExecStart=sh -c "rw & sleep 30 && ro & exec /usr/bin/zerotier-one"
```

4. After pasting the above code, save and exit the editor.
5. Reload the systemd daemon to apply these changes with the command `systemctl daemon-reload`.

Now Zerotier should be able to start up automatically at boot time.

laduke, 2023-12-20:
Is there some writable tmpfs or something directory zerotier can use? 

It writes some things to /var/lib/zerotier-one:
- only the first time it starts:  `authtoken.secret`, `identity.secret` for example 
- on each start: zerotier-one.pid zerotier-one.port (this is probably actually what is causing problems on boot)
- Some config files: whenever you... change config (leave/join networks)
- Some caches/logs: all the time, but these can be considered ephemeral and don't need to be saved across reboots.

mdevaev, 2023-12-20:
https://docs.pikvm.org/pst/

wzqvip, 2023-12-27:
Interesting. I wonder if there's future support.