pikvm/pikvm #636: update docs to explain jumper status

ykuksenko, 2022-02-03:
**Is your feature request related to a problem? Please describe.**
I am a new user and found the documentation lacking info about what the jumpers do. Specifically what state (shorted/open) does what.

**Describe the solution you'd like**
I would like the file docs/v3.md to be updated in the `IO ports and jumpers` section, specifically the 3 jumpers. Please add something like: 
```
2. HDMI reset jumper (Shorted: will ignore reset signal from HDMI input. Open: will restart MJPEG streaming service, when reset signal received)
4. Audio Capture jumpers (...)
14. HDMI backpowering jumper (Shorted: diode will stop current from HDMI input. Open: will allow current from remote HDMI device)
  - If your Pi does not turn on (no rainbow screen) when HDMI input is connected removing the jumper should fix this
```

**Describe alternatives you've considered**
Searching discord for this info. I sort of found my info but still don't really understand what state does what from the docs.

**Additional context**
The power on/rainbow screen thing was a real issue for me. Maybe that should go to known issues?
I do not know about the other jumpers.

arch1mede, 2022-02-05:
My 2 cents,

I see no value in adding these as they serve no purpose,
The only one of value is the HDMI backpowering jumper, which is documented already in the v3 document

*The power on/rainbow screen thing was a real issue for me. Maybe that should go to known issues?*
Which rainbow? Boot up rainbow? Display rainbow? None of the jumpers effect whichever rainbow you are mentioning.

ykuksenko, 2022-02-06:
Are the jumpers that serve no purpose just not implemented yet? (https://www.pishop.us/product/pikvm-v3-hat-for-raspberry-pi-4/ - implies audio may be a future feature).

Since I was affected by the HDMI backpowering issue. Is there any downside to always having the HDMI backpowering jumper removed? Particularly if I am changing input devices often. Item 6 on https://docs.pikvm.org/v3/#basic-setup does not say anything about that.

By rainbow screen I was referencing the initial screen a Pi 4 shows when it is powered on. I do not know about any other rainbow screens with Raspberry Pis. In my case the Pi was not able to turn on, hence no rainbow screen. Removing the HDMI backpower jumper fixed the issue.

arch1mede, 2022-02-07:
*Are the jumpers that serve no purpose just not implemented yet?*
The jumpers are not relevant till they are needed, when they are, they are documented like the hdmi backpowering

*Since I was affected by the HDMI backpowering issue. Is there any downside to always having the HDMI backpowering jumper removed?*
No.

*By rainbow screen I was referencing the initial screen a Pi 4 shows when it is powered on*
This is a RPI only screen and happens preboot and is well documented on the RPI forums.

ykuksenko, 2022-02-07:
Thanks @arch1mede.
I think this can be closed.

mdevaev, 2022-04-15:
Done!

commit 136c76a3 (2022-04-15): fixed #636: jumpers explanation
closed 2022-04-15 by mdevaev