pikvm/pikvm #638: Source files for PiKVM v3 ATX mounting brackets for 3D printing

denschub, 2022-02-07:
**Is your feature request related to a problem? Please describe.**
This is technically a "documentation feature request"?

I'd like to make some changes to the PCI mounts provided in https://docs.pikvm.org/stl/atx/ to make them easier to print and a bit less fragile. Unfortunately, working with the exported STL files is a bit hard, as it only contains some high-density mesh, but not the actual sources.

**Describe the solution you'd like**
If possible, I'd love to have the source files from whatever program this was created, or some neutral file formats like .step for example.

benmprojects, 2022-04-13:
Please upload and people can test

mdevaev, 2022-04-14:
There is a certain problem. We drew them in a hurry and now the sources of the 3D parts have been lost :/

denschub, 2022-04-14:
Well, too bad. :) I'll design my own eventually, and I'll share the source files here. No time estimate, tho.

mdevaev, 2022-05-19:
I'm closing it due inactivity (temporary), and will open it later if needed.

closed 2022-05-19 by mdevaev

fletchto99, 2022-07-06:
@denschub did you manage to design your own? I was facing the same issues when trying to print the STL files

denschub, 2022-07-06:
I'm currently using some brackets I hacked together using stamped metal, since I wasn't yet able to spend the time to design a clean, printable, version.

denschub, 2023-04-22:
As an official update since I opened this issue, and my previous message contained the word "yet": I have no plans of designing a more stable 3D-printable ATX mount in the future.

The reason here is simple: my v3 PiKVMs will "soon" be degraded into mobile use-cases where I don't need the ATX control, and for my two stationary machines that have the ATX board, they'll be replaced by PiKVM v4's, which come with PCI brackets that look like sheet metal.

If someone else wants to tackle this project in the future, note that this might not be feasible in the standard way. No matter how much you reinforce the "main part" of the bracket, there's always the lower lip where the bracket slides besides the mainboard into the case. This will always be the weak point, and if you print with PETG, it'll flex too much; and if you print with PLA, it'll shatter due to stress.

One might be able to get away with Carbon-reinforced PA11 or something, although I haven't tried. A more realistic workaround is probably to design some kind of external-bracket, where you have two clamps outside the case, that pull the mounting bracket outwards - ideally right next to the RJ45 socket. That way, when you push a cable into the board, it won't have a chance to deflect inwards. Good luck! :)