pikvm/pikvm #648: PiKVM v2 Pi Zero (Legacy) pacman database EoL

Karter87, 2022-02-12:
The Archlinux image for pi zero w contains obsolete pacman mirror link:
de3.mirror.archlinuxarm.org
ARMv6H is obsolete on this repository

**Expected behavior**
pacman -Syu tailscale-pikvm command to install tailscale

**PiKVM info:**
  - PiZero W
  - Linux pikvm 5.10.52-13-ARCH
  - Image: https://files.pikvm.org/images/v2-hdmi-zerow-latest.img.xz

**Additional context**
Changing /etc/pacman/.d/mirrorlist content from 
**Server = http://de3.mirror.archlinuxarm.org/$arch/$repo**
to 
**Server = http://alaa.ad24.cz/repos/2022/02/06/$arch/$repo**
Fix the problem (for now)

arch1mede, 2022-02-12:
This is not a legit fix, per the latest #news item that's on discord, zerow will be depreciated and 1 last image will be made for it then it will not recieve any more updates till the Rasbian variant comes out. However, this will only be a matter of time before its also deprecated.

Pointing to a non arch repo is a security risk.

closed 2022-02-12 by mdevaev

davidlang42, 2022-03-14:
This is now a big issue because the included kernel `5.10.23-2-ARCH` is vulnerable to the dirty pipe exploit `CVE-2022-0847`. Does anyone have any suggestions?