pikvm/pikvm #664: DHCP Hostname not sent - helpful for local DDNS

lutzky, 2022-03-03:
**Describe the bug**

Many networking systems provide local dynamic DNS for any client that sends a DHCP hostname. Annoyingly, in many setups this is the *only* way to do it (e.g. if you're using just a Unifi Dream Machine and don't want to modify its internals). This can be accomplished in pikvm like so (example for wireless on Raspberry Pi Zero W) - create `/etc/dhclient.conf` with the following contents:

```
interface "wlan0" {
  send host-name "pikvm";
}
```

After rebooting pikvm, running the following command on another machine in my network gives this:

```
$ host pikvm
pikvm.localdomain has address x.x.x.x
```

It works - but:

1. Is undocumented in PiKVM's documentation
2. Feels hacky

If this does not seem hacky to you, I think a reasonable fix for the bug is adding documentation about this to pikvm's docs.

arch1mede, 2022-03-04:
My 2 cents, this is not PiKVM related, this is a arch linux related issue, this is the reason why its not documented. Everyone's network is different so this wouldn't make sense to add it to the image repo.

arch1mede, 2022-03-04:
To add further context to that not all networks are the same, I myself do not have a /etc/dhclient.conf but I am able to get an output from host pikvm.

lutzky, 2022-03-04:
While that is true, https://docs.pikvm.org/wifi/ and https://docs.pikvm.org/tailscale/ are two examples of things that are "not PiKVM related" but are indeed useful documentation.

arch1mede, 2022-03-05:
> While that is true, https://docs.pikvm.org/wifi/ and https://docs.pikvm.org/tailscale/ are two examples of things that are "not PiKVM related" but are indeed useful documentation.

Yup, this is true. I will of course let the dev make the final call, I was just providing my 2 cents before a final decision is made. Again, I would point out that not all networks are built the same as I do not have this same issue.

At some point a line needs to be drawn between PiKVM and arch linux support though.

mdevaev, 2022-03-11:
PiKVM uses systemd-networkd to configure ethernet and this option [is enabled by default](https://www.freedesktop.org/software/systemd/man/systemd.network.html#SendHostname=). If this doesn't work in the new images, then either it's a server problem or systemd. In any case, from our side, I don't see how this can be fixed (except using dhclient).

closed 2022-03-11 by mdevaev

lutzky, 2022-03-11:
Indeed, I believe I'm on an older image, where systemd-networkd was used for ethernet and not for wireless ( am on a Pi ZeroW). Indeed, the new documentation seems to instruct to use systemd-networkd for wifi as well, so this bug is a moot point. Thank you!