pikvm/pikvm #665: Mic input

edescourtis, 2022-03-05:
**Is your feature request related to a problem? Please describe.**
I have a locked-down machine from my employer that I would like to control from my laptop. I sometimes require two-way audio. That means I need my mic on my laptop to be captured and sent to the computer via its MIC input port.

**Describe the solution you'd like**
The quality of the audio is not a big issue for me. I want to be able to use MIC audio. That means we need a way to send sound to the MIC port.

**Describe alternatives you've considered**
No other options exist; I've looked everywhere.

**Additional context**
Maybe a way to turn on and off the MIC might be a good idea.

arch1mede, 2022-03-05:
Hi, sound is not supported at this time, it is on the roadmap though. 

https://docs.pikvm.org/faq/

[image]

mdevaev, 2022-03-11:
Sup. Audio, including two-way, will be made in the future: #97

closed 2022-03-11 by mdevaev