pikvm/pikvm #669: Offical(?) Rack Mount for the project

SchoolGuy, 2022-03-06:
**Is your feature request related to a problem? Please describe.**

Currently I see no way of mounting this to a rack cleanly. There are rack-mount solutions for for Raspberry Pi's, however the ports and sizing of the enclosures is mostly off.

**Describe the solution you'd like**

A solution in the docs to mount this appliance into the rack. Ideally with a "supported" or at least recommended KVM switch.

**Describe alternatives you've considered**

There are 1U open-top cases from various places but the cable management is at best acceptable to my standards.

**Additional context**

Nothing more comes into my mind I could put here.

arch1mede, 2022-03-08:
Hi, [this](https://www.thingiverse.com/thing:4968905) is closest to what you are looking for. There is nothing official at this moment, maybe when V4 is released they will have something official. Till then its mostly all DIY.

We cannot recommend any KVM switch as its not part of the PiKVM ecosystem but the one that has the least amount of issues is the [EZCOO](https://docs.pikvm.org/multiport/) 4 port.

SchoolGuy, 2022-03-08:
@arch1mede That is exactly what I was looking for! Thanks a lot. If the maintainers want to keep it open, let it stay open, otherwise feel free to close this ticket.

closed 2022-03-11 by mdevaev