pikvm/pikvm #672: WebUSB

stevenruidigao, 2022-03-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
I'd like to be able to connect USB devices over the browser to the machine controlled by PiKVM

**Describe the solution you'd like**
This might be a lot of work, but is it possible to use WebUSB to get USB devices to communicate through the browser to the machine on the other end?

**Describe alternatives you've considered**
I don't really know of any alternatives right now.

**Additional context**
I know that Arduinos are capable of attaching to the Arduino Create web IDE. Is there anything preventing this from working for PiKVM?

arch1mede, 2022-03-14:
You mean like VirtualHere?

stevenruidigao, 2022-03-14:
Yeah, kind of like that, except through the browser, and there's no software on the controlled machine, just a PiKVM.

mdevaev, 2022-04-30:
It's really hard, but possible I think.

mdevaev, 2022-07-26:
Now it will be a generic task for USB passthrough.

PrasadKCode, 2022-09-01:
> Yeah, kind of like that, except through the browser, and there's no software on the controlled machine, just a PiKVM.

Hi, did you find any other alternative for this need? Can you please let me know?

mdevaev, 2022-10-04:
> Any updates here?

No, sorry, it's a big and complex task

peterasorensen, 2022-10-04:
Can you share where you're at + any research? Maybe can lend a hand

mdevaev, 2022-10-04:
At the moment there is nothing useful, we are discussing internally how this can be done for libcomposite on the fly.

leonmeijer, 2023-01-24:
Would like to see this as well. Scenario: Connect my PC with smart card reader to the controlled device.

joeknock90, 2023-02-06:
> Would like to see this as well. Scenario: Connect my PC with smart card reader to the controlled device.

I just setup my piKVM for work and this is exactly what I was looking to do too!

SilentHunter124, 2023-05-02:
Is there any progress on that, I think there was a post about this scenario on reddit but can't find it anymore? I just started looking around and I think WebUSB will not be enough... maybe this one is needed: https://usbip.sourceforge.net/ ? Is anybody experimenting with such software packages?

carlosmax3D, 2023-07-08:
I did some research; for example in the scenario of a yubikey. It acts like a keyboard that send a generated string, in theory if we can emulate another keyboard with the same device IDs of our yubikey, we can just paste the string through pikvm and bypass the yubikey request, what do you think?, it is possible??

mdevaev, 2023-07-08:
Probably yes

SilentHunter124, 2023-07-08:
This is mostly not the use case, Smartcards don‘t act like a keyboard for example. Headsets also not…

carlosmax3D, 2023-07-25:
What about this solution? https://github.com/luigifcruz/webusb-libusb, Looks promising, the only issue is that only works in chromium based browsers.

mdevaev, 2023-07-27:
We still need to handle it on raspberry's side. It's hard.

spoelstraethan, 2023-08-05:
I think calling this "WebUSB" is going to lead everybody down the wrong path. What you basically want is USB/IP (USB over IP), but you want to run the "server" where the device is attached and make the remotely managed PC a "client" by tunneling through PiKVM.

https://docs.kernel.org/usb/usbip_protocol.html
https://www.kernel.org/doc/readme/tools-usb-usbip-README

See also [VirtualHere](virtualhere.com) and the WSL docs on doing  a USB device from Windows into the WSL vm space. https://learn.microsoft.com/en-us/windows/wsl/connect-usb

SilentHunter124, 2023-08-06:
@spoelstraethan You are absolutely right except one thing: VirtualHere and other existing solutions on the market use the wrong DIRECTION. By using USB over IP, VirtualHere and other solutions require a server which has already all needed USB devices attached somewhere in your network AND more critical path, the target device (which would be the one attached to pikvm) is fetching those devices by an installed software client. Normally the target system attached to pikvm is not able to install any software as a client for consuming it. This is a show stopper.

I actually like your idea of registering a new USB Device (maybe just reserve one channel for this use case) on the pikvm hardware and then tunnel the device from webusb to this over ip? Would that technically work?

As a user you could grab any device attached to the client by the chrome/chromium plugin webusb like you would do in RDP sessions?

I would love to try and error some solutions but I have actually no idea what steps to do to get this working (as workaround by hand or something like that) :(

KarlRW, 2024-02-27:
+1
Looking to emulate webcam on remote device using local webcam if possible.

spoelstraethan, 2024-02-27:
I think you'll need to handle the webcam with a separate Pi+camera and some
way of network tunneling because the CSI port on the PiKVM is already in
use by the HDMI to CSI bridge.

Check out the Raspberry Pi blog and showmewebcam

https://www.raspberrypi.com/news/how-to-build-your-own-raspberry-pi-webcam/

connorzk, 2025-07-01:
any update here?  this is my most wanted feature .

because the target pc need an usb smart card to login.