pikvm/pikvm #674: Cannot update installed pi zero w

sharicov, 2022-03-13:
When running update command **pacman -Syy** getting following error:

```
:: Synchronizing package databases...
 core.db failed to download
error: failed retrieving file 'core.db' from de3.mirror.archlinuxarm.org : The requested URL returned error: 404
error: failed retrieving file 'core.db' from mirror.rackspace.com : The requested URL returned error: 404
error: failed retrieving file 'core.db' from mirror.rackspace.com : The requested URL returned error: 404
error: failed to synchronize all databases (failed to retrieve some files)
```

Didn't use this device for quite a long time.

mdevaev, 2022-03-13:
Sup. Please read here: https://www.reddit.com/r/pikvm/comments/sq6e63/zero_w_1_and_rpi_1_endoflife/

closed 2022-03-13 by mdevaev

si458, 2023-06-15:
https://forums.raspberrypi.com/viewtopic.php?t=326235#p2005629
this will help with arch pacman packages but wont help with pikvm packages sadly