pikvm/pikvm #675: Issues Starting TailScale

Bruce0286, 2022-03-13:
New to Pi and PiKVM (and relatively low experience in ArchLinux) so please excuse me in advance if I ask for additional clarification to any techie bits! Built my PiKVM with the V3 hat and it's incredible so thank you! I was hoping to get tailscale running for simple and secure access away from home. I have installed the tailscale package as per the instructions on the PiKVM page.

# rw
# pacman -Syy
# pacman -S tailscale-pikvm
# systemctl enable --now tailscaled
# tailscale up

The initial commands are all successful but when running systemctl enable --now tailscaled I receive the following error:

Job for tailscaled.service failed because the control process exited with error code.
See "systemctl status tailscaled.service" and "journalctl -xeu tailscaled.service" for details.

systemctl status tailscaled.service returns:

x tailscaled.service - Tailscale node agent
     Loaded: loaded (/usr/lib/systemd/system/tailscaled.service; enabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Sun 2022-03-13 16:31:19 MSK; 7min ago
       Docs: https://tailscale.com/kb/
    Process: 1335 ExecStartPre=/usr/sbin/tailscaled --cleanup (code=exited, status=1/FAILURE)
    Process: 1336 ExecStopPost=/usr/sbin/tailscaled --cleanup (code=exited, status=1/FAILURE)
        CPU: 10ms

Mar 13 16:31:19 pikvm systemd[1]: tailscaled.service: Scheduled restart job, restart counter is at 5.
Mar 13 16:31:19 pikvm systemd[1]: Stopped Tailscale node agent.
Mar 13 16:31:19 pikvm systemd[1]: tailscaled.service: Start request repeated too quickly.
Mar 13 16:31:19 pikvm systemd[1]: tailscaled.service: Failed with result 'exit-code'.
Mar 13 16:31:19 pikvm systemd[1]: Failed to start Tailscale node agent.

I am using a RPi 4 2GB RAM and the latest Hat package from the PiKVM website. Tailscale package versions are:

tailscale-pikvm-0.4-1 and tailscale-1.22.1-1. I don't know if this is an issue with newer versions of the tailscale package or if anyone else can reproduce this?

Thanks for your assistance.

arch1mede, 2022-03-14:
Tailscale is a 3rd part app, the issues you are experiencing is at the OS level which isn't pikvm related. My suggestion would be to reflash your sd card with the latest image for your platform and do this again, I just ran these commands and it was fine.

mdevaev, 2022-03-16:
I've just tested it and everything is working. Try to update full os.

Bruce0286, 2022-03-16:
Thank you both for taking the time to try it. It was my reading skills at fault and I had missed the syu to do a full update during the setup of the pikvm where you have clearly put "this resolves most issues" yet somehow I skipped over that step. All working great now thank you both.

closed 2022-03-16 by mdevaev