pikvm/pikvm #677: Cannot get video if using POE HAT+

ocalvo, 2022-03-14:
**Describe the bug**
When using a HDMI-CSI module and an official raspberry PoE HAT+, the HDMI signal is not detected.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Use a Raspberry Pi4, HDMI-CSI module and PoE HAT+
2. Flash latest release.

**Expected behavior**
HDMI input can be seem

**Screenshots**

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 11
 - Browser [e.g. chrome, safari] Edge
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] RPi4
  - PiKVM platform [e.g. v2-hdmi]
  - Video capture type [e.g. CSI bridge] CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.48-1
kvmd-fan 0.1-1
kvmd-oled 0.8-1
kvmd-platform-v2-hdmi-rpi4 3.48-1
kvmd-webterm 0.41-1
  - uStreamer version: `pacman -Q | grep ustreamer`
ustreamer 4.9-1
  - Linux kernel: `uname -a`
Linux pikvm 5.10.52-13-ARCH #1 SMP Sat Nov 20 09:40:57 MSK 2021 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

ocalvo, 2022-03-14:
Removing the PoE HAT+ and powering via USB-C makes the issue go away.

mdevaev, 2022-03-15:
Hello. See here: https://docs.pikvm.org/faq/#video-problems
[image]

closed 2022-03-15 by mdevaev