pikvm/pikvm #678: Seeing constant rtc0/sinc_epoch errors in logs?

hewfish, 2022-03-15:
**Describe the bug**
In the PiKVM there are constant rtc0/since_epoch errors about no such file or directory

**To Reproduce**
Steps to reproduce the behavior, like:
1. Go to 'log viewer from PiKVM GUI'

**Expected behavior**
no errors?

**Screenshots**
[2022-03-15 09:15:46 kvmd-watchdog.service] --- kvmd.apps.watchdog                INFO --- Running watchdog loop on RTC0 ...
[2022-03-15 09:15:46 kvmd-watchdog.service] --- Traceback (most recent call last):
[2022-03-15 09:15:46 kvmd-watchdog.service] ---   File "/usr/bin/kvmd-watchdog", line 9, in <module>
[2022-03-15 09:15:46 kvmd-watchdog.service] ---     main()
[2022-03-15 09:15:46 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 119, in main
[2022-03-15 09:15:46 kvmd-watchdog.service] ---     options.cmd(config.watchdog)
[2022-03-15 09:15:46 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 80, in _cmd_run
[2022-03-15 09:15:46 kvmd-watchdog.service] ---     _reset_alarm(config.rtc, config.timeout)
[2022-03-15 09:15:46 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 60, in _reset_alarm
[2022-03-15 09:15:46 kvmd-watchdog.service] ---     now = _read_int(rtc, "since_epoch")
[2022-03-15 09:15:46 kvmd-watchdog.service] ---   File "/usr/lib/python3.10/site-packages/kvmd/apps/watchdog/__init__.py", line 50, in _read_int
[2022-03-15 09:15:46 kvmd-watchdog.service] ---     with open(_join_rtc(rtc, key)) as value_file:
[2022-03-15 09:15:46 kvmd-watchdog.service] --- FileNotFoundError: [Errno 2] No such file or directory: '/sys/class/rtc/rtc0/since_epoch'
[2022-03-15 09:15:50 kvmd-watchdog.service] --- kvmd.apps.watchdog                INFO --- Running watchdog loop on RTC0 ...

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 99.0.4844.51 (Official Build) (64-bit)

**PiKVM info:**
  - Raspberry Pi board: CM4 on BLIKVM (v3 HAT)
  - Video capture type: WebRTC / 
  - KVMD version: `pacman -Q | grep kvmd`
       kvmd 3.56-1
       kvmd-fan 0.9-1
       kvmd-oled 0.12-1
       kvmd-platform-v3-hdmi-rpi4 3.56-1
       kvmd-webterm 0.41-1
  - uStreamer version: `pacman -Q | grep ustreamer`
       ustreamer 4.13-1
  - Linux kernel: `uname -a`
       Linux pikvm 5.10.52-14-ARCH #1 SMP Thu Dec 2 09:40:42 MSK 2021 armv7l GNU/Linux

**Additional context**
Just a kvmd-watchdog.service repeating error in the logs

arch1mede, 2022-03-15:
We only support DIY and prebuild (preconfigured) images for PiKVM V3 hardware.

We have no CM4 hardware, provide no instructions to build some and we don't spend time to support features we don't use.

Support for third-party hardware rests solely on the shoulders of the party selling it AKA BLIKVM devs. We won't, and we don't get a penny to provide support for those functions. Thus, we don't support third-party hardware.

arch1mede, 2022-03-15:
Raspberry Pi board: CM4 on BLIKVM (v3 HAT) <-- This is incorrect, V3 HAT is NOT the same as this 3rd party hardware. Please file this under the proper dev github issues (BLIKVM).