pikvm/pikvm #679: Add 2FA authentication method

jobggun, 2022-03-18:
**Is your feature request related to a problem? Please describe.**
When I use PIKVM outside, I am worried about the password being leaked. (unsafe network, cctv, etc.)

**Describe the solution you'd like**
Adding login method with password and [Time-based One-time Password Algorithm](https://en.wikipedia.org/wiki/Time-based_One-time_Password_Algorithm) (TOTP; specified in RFC 6238)

arch1mede, 2022-03-18:
I would install Tailscale if you are going to expose this on the inet.

mdevaev, 2023-01-22:
https://www.geeksforgeeks.org/two-factor-authentication-using-google-authenticator-in-python/

mdevaev, 2023-01-23:
Done: https://docs.pikvm.org/auth/#two-factor-authentication

closed 2023-01-23 by mdevaev