pikvm/pikvm #683: Persistent Partition

djarbz, 2022-03-21:
I'm not sure how exactly your Read-Only filesystem works and I am interested in creating a second partition on my SD card that would be persistent.
This would allow me to run additional services on my Pi such as Ansible to create the "ultimate" home lab management platform.

Would there be any caveats to my idea that I am unaware of in regards to your ro/rw mechanism?

arch1mede, 2022-03-22:
@djarbz No but its not recommended that you add things to your pikvm. In fact what you are asking for has very little to no relationship to pikvm. You would need to review the Arch Wiki, edit /etc/fstab and maybe look in the [faq](https://docs.pikvm.org/faq/#first-steps) to give you an idea on what to do next.

djarbz, 2022-03-29:
@mdevaev I see that there are 3 partitions on my SDCard (32GB).

```
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p1  256M   42M  215M  17% /boot
/dev/mmcblk0p2  5.6G  3.0G  2.3G  57% /
/dev/mmcblk0p3   24G   32K   24G   1% /var/lib/kvmd/msd
```

Can you tell me what `/var/lib/kvmd/msd` is used for?
Looks like it is for the ISO files we upload?
Can you advise if there would be any issues for me to change the fstab entry for this partition to remove the `ro` flag?
This way I can store my persistent data here.

djarbz, 2022-03-29:
I decided to give it a try by removing the `ro` flag so my fstab entry would read as follows, but it still mounted as read only...?

```
/dev/mmcblk0p3 /var/lib/kvmd/msd  ext4  nodev,nosuid,noexec,errors=remount-ro,data=journal,X-kvmd.otgmsd-root=/var/lib/kvmd/msd,X-kvmd.otgmsd-user=kvmd  0 0
```

mdevaev, 2022-03-30:
@djarbz also edit /boot/cmdline.txt and replace `ro` to `rw` there.

djarbz, 2022-03-31:
Thanks @mdevaev I ended up just putting my RW files on a high endurance USB drive.

mdevaev, 2022-06-20:
Done: https://docs.pikvm.org/pst

closed 2022-06-20 by mdevaev