pikvm/pikvm #687: Virtual Printer

myMartek, 2022-03-29:
I would really love to have a virtual printer in PiKVM that attaches to the guest.

In general: It should be a "print to pdf" virtual printer that is available on the guest as a standard printing device and would output a PDF that is downloadable via PiKVMs Webpage.

Other possibilities would be to use a virtual flash drive that I print with the guest to and then attach it to the PiKVM host. But that would be very complex and would take very long.

mdevaev, 2022-03-29:
I think the virtual printer is a very special case and the easiest way is a virtual flash. Both will take some time to implement.

mdevaev, 2022-07-26:
Okay, after thinking about it. A virtual printer is too specific a thing to support. Considering that PDF printing is a standard feature of any software, it is much more important to be able to download a file from the host. Since this is also mentioned here as a possible alternative, we already have a more general task on this topic: #612

I will close the current one as a duplicate, so keep an eye on that task.

closed 2022-07-26 by mdevaev