pikvm/pikvm #688: How can i run PiKVM in proxmox?

connected201, 2022-03-30:
hi, i want to run pikvm in proxmox not on raspberry, it's possible to install pikvm over simple desktop arch distro?.
I tried with balena etcher write pikvm image, but pikvm not starting, also i wrote that if i want to run in qemu i need to compile with  versatile.
(`https://qemu.readthedocs.io/en/latest/system/arm/versatile.html`).
And after will be avaialbe to run comand like 
`qemu-system-x86_64 versatilepb -cpu arm1176 -hda raspbian.img -kernel kernel-qemu -m 1024 -append "root=/dev/sda2"`
to launch raspbian image.
Maybe exist more simplest way to launch pikvm without raspberry?

connected201, 2022-03-30:
upd, i found this git repo [`https://github.com/dhruvvyas90/qemu-rpi-kernel`](https://github.com/dhruvvyas90/qemu-rpi-kernel)

i will try this solution and reply about my success.

arch1mede, 2022-03-30:
**_hi, i want to run pikvm in proxmox not on raspberry_**
For what reason?

**_it's possible to install pikvm over simple desktop arch distro?._**
Not easily

**_to launch raspbian image._**
Rasbian? This is running arch arm linux

**_Upd. it's look like this kernel is only for raspbian?_**
Rasbian? This is running arch arm linux

arch1mede, 2022-03-31:
Hi, this is an unsupported configuration, I would suggest you get arch arm image working in proxmox first (Why you would need this I have no idea) Then you should be able to to switch to PiKVM however it will not run OOB as its looking for some capture device + msd partition.

closed 2022-03-31 by mdevaev

viniciusvillas, 2023-04-03:
I'd just like to add my 2 cents by presenting a real use case:

I have a 19" rack at home with 3 servers, which are mainly kept by solar powered batteries. I'd like to control these machines remotely, and I could buy a Raspberry Pi and install PiKVM, but that would be another device to keep and power 24x7, thus, not being an interesting option.

One of these servers is already running Proxmox and it would be *very* interesting if I could run PiKVM inside it, and then connect it to the other machines.

I understand this type of installation is not supported. I just wanted to share how I intended to use it, IF it was supported.

Cheers!

arch1mede, 2023-04-04:
There are just too many things to contend with, PiKVM is a hardware+software solution, both are necessary to function properly:
PC's do not have an OTG port, so you need to add in an arduino (external)
You'll then need to play with the partitions because you'll need an MSD partition
You'll need to add a capture device (USB) (external)
KVM switch (external)

You have 3 servers running, adding a PiKVM would hardly feel a difference in the power draw, its likely the power draw of spinning up a new VM+adding all of the hardware to get it working is the same as adding a stand alone PiKVM. This would be no different if you added a Raritan or Lan Spider to get KVM functionality.

https://github.com/purplepalmdash/pikvmx86 you can see that someone attempted something similar here but hasn't been updated in 3 years, most likely due to too much time and effort to keep it updated. 

It's better if you bought a USB crash cart like what StarTech offers, run a windows VM, pass through the USB to activate the USB Crash cart + KVM switch(external). This is 2x the price, FYI.

At the end of the day, this is why its unsupported, too many things can go wrong, where to get support for it? A PiKVM is pretty much a set it and forget it IoT, especially more so if you get a pre-assembled version and you get support for it either here or through Discord.

wbrione, 2024-02-19:
Ladies and gentlemen, sorry for raising this subject again, but as it is so pertinent, I am going through a similar situation, where, if possible, it would be interesting to have **PiKVM** running in a containerized environment, dispensing with need to maintain exclusive hardware for this activity. I understand that, if the software solution was designed to run specifically on a Raspberry, in the availability of solutions such as **PiMox** (Promox over Raspberry), where it supports **LXC containers**, I believe that if somehow, the PiKVM image could be converted to the "template" format (tar.xz), with the proper association of some devices present on the host system, it might be possible to run its services. Anyway, I'm still in the preliminary study phase, but I'm putting this scenario out there for the "enthusiasts" on duty who have a more in-depth knowledge of the subject ;-)

Now, already answering the possible question of why I want to enter this scenario, I honestly think it is a waste to have hardware like a Raspberry Pi 4/5 allocated exclusively for a single service, in this case, PiKVM. For example, I installed Proxmox on my Pi4, where I have an LXC container running specifically for Docker, where in this, I have several services running, with extremely low resource consumption, so why not have another LXC container, exclusive for the PiKVM?! To me, it makes perfect sense, even more so taking into account all the facilities included in this wonderful piece of software, which is Proxmox ;-)

Well, now, for those who want to know a little more about how to install Proxmox on a Raspberry, you should already know the PiMox project ([https://github.com/pimox/pimox7](https://github.com/pimox/pimox7)), but there are still other possibilities, such as I applied it in my environment, enabling the installation of version 8 onwards of Proxmox ([https://github.com/jiangcuo/Proxmox-Port](https://github.com/jiangcuo/Proxmox-Port))

Anyway, remember, I know that in a capitalist world, everyone needs money to live, but, when it comes to free software, sometimes what motivates developers is curiosity and the desire to go further, to make things happen ;-)

Have a great week everyone, and see you next time.

arch1mede, 2024-02-19:
Hi, At this time, the PiKVM Team does not have any plans to design out of scope implementations of PiKVM. If you somehow figure it out, please share so that the community can benefit.

anipad, 2025-09-01:
> Ladies and gentlemen, sorry for raising this subject again, but as it is so pertinent, I am going through a similar situation, where, if possible, it would be interesting to have **PiKVM** running [...]

Hi, this is a long shot given the time passing, but did you somehow manage to get it to work? I am also very much intereset in PiKVM running virtually in a Proxmox on Arm environment (VM or LXC) with working HDMI to USB and USB to USB passthrough.