pikvm/pikvm #689: Trying to enter BIOS does not work and causes keyboard problems

cjmcgee, 2022-03-30:
v3.3 HAT with a Pi 4b, latest software as of 3/30/22.  Connecting via Chrome (latest) on Windows 10. 

I was trying to enter the BIOS by spamming the DEL key in the web interface while the computer rebooted. It would not enter the BIOS and after the monitored computer booted up into the OS (Windows 11) the keyboard directly connected (USB) to this computer would not work.  Rebooting the PiKVM made the local keyboard work again.  

Steps to reproduce:
1. Login to the PiKVM web interface and open KVM mode
2. Reboot the monitored computer
3. Start hitting the DEL key repeatedly as soon as the screen goes black
4. See the POST screen which says "Hit DEL to enter the BIOS" 
5. BIOS is not entered
6. See the OS fully boot up
7. Try to enter a password using the local keyboard and see that the local keyboard is not working
8. Reboot the PiKVM and see that the local keyboard now works again

I am guessing the problem is that I am sending keyboard command while the USB is in a powered off or connecting state. Sending USB  commands in these states probably generates errors and thus gets the PiKVM software into a bad state; a bad enough state that it is causing problems on the USB blocking the local keyboard from working.  

If I could manage to hit DEL only while the POST screen showed (about 1 sec) and USB was fully initialized and not before (while it was rebooting) then I am guessing it may not have this problem. The problem is that I have an extremely short POST window so I have to spam the key as it boots up to get the key press in during the few milliseconds between where the computer actually recognizes the keyboard commands and when the OS begins to boot.

arch1mede, 2022-03-30:
Which BIOS for which mother board? Not all BIOS's are compatible, you can try and disable MSD and see if that resolves your issue or you might need a arduino. ¯\_(ツ)_/¯

This isn't really an issue, just a known behavior with some BIOS's.

cjmcgee, 2022-03-30:
MSI x570-a pro, BIOS 7C37vHF3.  

I am not really understanding the MSD issue here; could you explain this known behavior?  

For more information, the PiKVM seems to work perfectly all other times other than this specific scenario. If I do not try to enter the BIOS then everything works perfectly; the local KB work fine, the remote KB works fine, the MSD works fine, etc. Getting into the BIOS with the local USB keyboard also works fine.  

I disabled MSD and reboot and I was able to get into the BIOS this time (which hopefully is not just a timing thing). But the problem I seem to have now is that if I exit the BIOS (without saving changes) it gets stuck on a black screen and does not boot into the OS. I tested this a couple times and it was consistent. I saw it get stuck on a black screen a couple times when testing out trying to get into the BIOS before; sequence was a) try to get into BIOS, b) KB does not work, c) reboot, d) stuck at black screen.  It did not have any issue with this without the PiKVM connected. A reset seems to fix it; so it is not the end of the world, but it is a bit worrying.

arch1mede, 2022-03-30:
I have redacted my previous statement, the devs MIGHT have a fix for this however I cannot say if it will fully fix your issue.

NiKiZe, 2022-03-30:
Keyboards can work in different ways, especially in the early boot process. There is also that there is multiple functions in the same device, which might not work with some hosts. (Or hosts software implementations)

With less usb features enabled some of the incompabilities can be reduced.

arch1mede, 2022-04-05:
@cjmcgee Did you try this again with the newest [version](https://discord.com/channels/580094191938437144/712960053174599712/959488146817568800)?

cjmcgee, 2022-04-06:
It looks like the original problem is fixed.  I updated to the latest software then re-enabled MSD.  I then rebooted and spammed the DEL key and was able to get into the BIOS.  The remote keyboard worked in the BIOS. When I exited the BIOS without saving changes I again got stuck on a black screen, so that weirdness is still there; but using the "Click reset" option was able to get me past this. When windows booted up both the local and remote keyboard worked correctly. 

I think this issue is fixed for me now; closing the item.

closed 2022-04-06 by cjmcgee

mdevaev, 2022-04-06:
Thanks for reporting! :)