pikvm/pikvm #691: Adding support from DIY-IPMI for SainSmart 8-Channel Relay Module

Fmstrat, 2022-03-30:
Long time no chat, I'm the original author of https://github.com/Fmstrat/diy-ipmi for which I used to have credit on here as this was based off that, but as I'm updating my systems I realize pikvm has progressed WAY beyond where I am today, so I'm officially switching over. ;)

As I still have the SainSmart relays (https://www.amazon.com/dp/B0057OC5WK) for controlling PC power, I'm likely going to add support to `pikvm` so I can use them. The question:

- Is there interest in me providing a PR that does this officially, and if so, is there a specific path forward from a formatting/usage perspective you would prefer?

If so, I'll spend some time to make this formal. If not, I'll just do it in a hacky way that works for me in my particular use case.

Once I get spun up and assuming all goes smooth, I'll likely add a link to the top of my repo to direct people here, too.

mdevaev, 2022-03-30:
Hey, I'm glad to see you here :)

To add that switch yu need to write GPIO plugin (i.e. "pseudo gpio"). This way allow you to create control menu without touching web ui code. Here you can find full documentation, example for Ezcoo switch and the plugin code:
- https://docs.pikvm.org/gpio
- https://docs.pikvm.org/ezcoo
- https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/ugpio/ezcoo.py
I will accept your PR to the mainline ofc.

mdevaev, 2022-03-30:
Maybe the native `gpio` plugin would work with your relay module without modifications. Check the doc.

Fmstrat, 2022-03-31:
I'm also wondering if, given this as the default in `main.yaml`:
```
    atx:
        type: gpio
        power_led_pin: 24
        hdd_led_pin: 22
        power_switch_pin: 23
        reset_switch_pin: 27
```
And this as the wiring diagram for the relay switch: https://raw.githubusercontent.com/Fmstrat/diy-ipmi/master/Resources/Diagram.png, and this as pins: https://circuitdigest.com/sites/default/files/inlineimages/u3/Raspberry-Pi-Zero-GPIO-Pins.jpg

That I couldn't just do this:
- Pi 2 (+5V) to Switch 1 (+5V)
- Pi 6 (GND) to Switch 10 (GND)
- Pi 16 (GPIO23) to Switch 2 (Power relay, far left)
- Pi 13 (GPIO27) to Switch 3 (Reset relay, next to the right)

It looks like the `pulse` you're using (https://github.com/pikvm/kvmd/blob/ecb585866cd7d4082d71fd50555efa7b2029a3aa/kvmd/aiogp.py#L38) is doing something similar to my `output` (https://github.com/Fmstrat/diy-ipmi/blob/825b8658c458c93e0d8fe43b02077093d8cf4503/Pi3/rebootServer.py#L30).

If so, the only update required to make this usable is documentation I think.

mdevaev, 2022-03-31:
> Would these be acceptible alternatives to the omrons? https://www.farnell.com/datasheets/73758.pdf

Yes, but you should be careful with polarity for the button emulation. Motherboard pins for buttons actually do have polarity.

> It looks like the pulse you're using similar to my output

Yup.

> I'm also wondering if, given this as the default in main.yaml

You can use /etc/kvmd/override.yaml and redefine these parameters: https://docs.pikvm.org/first_steps/#configuring-pikvm

Fmstrat, 2022-04-02:
[image]

Works like a charm. No changes needed to any config or code.

Fmstrat, 2022-04-02:
I will make a little diagram, update the DIY doc, and submit a PR for review.

mdevaev, 2022-04-02:
Hehe :)

mdevaev, 2022-04-20:
Boop

mdevaev, 2022-04-30:
Sup?

mdevaev, 2022-05-02:
I'm closing this issue since there are actually no technical problems here. If you want to do PR or discuss something else, I will open it again :)

closed 2022-05-02 by mdevaev