pikvm/pikvm #696: Support of client-certificates  for web-Interface and public-key authentication for ssh by web-interface

Marsupilani, 2022-04-01:
Problem:
If there is a not known Firewall with HTTPS interception on the way between browser and pikvm it is possible to break the https encryption and catch the login credentials. (I will use the pikvm for first installation jobs in foreign environments.)
IPsec tunnel may be a alternative, but harder to manage (IPsec is often not allowed throw firewalls) 

Solution idea:
If it would be possible to registration an authorized client-certificate for authentication it is impossible to steal login credentials by that way (like public-key authentication with ssh) and in addition login will not work if HTTPS Interception is in the way, so you will be warned.

You can do it by editing web-server config and sshd config, but such a security config (with optional wizard to generate keys/certificate)) in the web-interface would be nice and easier to setup.

arch1mede, 2022-04-05:
Isnt this more arch arm linux than pikvm? PiKVM is just service running on top of arch arm linux using nginx.

mdevaev, 2022-04-07:
I need to think about it.

arch1mede, 2022-04-20:
My 2 to 5 cents, isn't this more arch arm than PiKVM therefor up to the end user to implement?

mdevaev, 2022-07-31:
Okay, I figured it out. It's too custom to support out of the box, but since Nginx is running at the top level of HTTPS in PiKVM, you can do it yourself. Just don't forget to check for Nginx configuration updates when you update the OS. Occasionally I add new endpoints or change existing ones.
- You can find Nginx configs in `/etc/kvmd/nginx/`
- Systemd service is `kvmd-nginx`, not just `nginx`
- Tutorial: https://fardog.io/blog/2017/12/30/client-side-certificate-authentication-with-nginx/

closed 2022-07-31 by mdevaev