pikvm/pikvm #697: How do we determine the current version and date of piKVM?

GitJJHub, 2022-04-01:
So that we can decide if we need to run the update or not.
Thanks,
Jerry

mdevaev, 2022-04-01:
Sup.
`pacman -Q | grep kvmd`
Also you can view it in the web ui. Click `System -> About` menu.

GitJJHub, 2022-04-01:
Thanks,
Not sure what is the “web ui”
This is what my web access looks like:
***@***.***

closed 2022-04-02 by mdevaev

GitJJHub, 2022-04-07:
Sorry to email directly but I can’t find the answer anywhere.
What is the purpose of the “Video Out” port?
I was hoping I could use it as an HDMI pass-through to a display attached to the computer being remotely controlled but I am also guessing that is not the purpose (though it would be great if that was offered).
Also, if there is a local keyboard and mouse attached to that computer, will they work while the piKVM is also attached?
It would be great if these answers appeared in the FAQ.
Thanks,
Jerry

I mean the browser interface.
[image]<https://user-images.githubusercontent.com/416259/161315316-7a9dd962-e711-49f4-8a2d-5de259ff926b.png>