pikvm/pikvm #699: Keyboard/Mouse icons are orange in the KVM web GUI. It shows "Keyboard captured inactive/busy" and "Mouse free inactive/busy"

Maximka101, 2022-04-04:
I recently bought and assembled PiKVM v3.3. All works exсept a virtual keyboard and a virtual mouse.
Keyboard/Mouse icons are orange in the KVM web GUI. It shows "Keyboard captured inactive/busy" and "Mouse free inactive/busy".
At first, I read FAQ and Community FAQ. Tried to change cables (USB C-USB A x4pcs.; USB A-USB A; USB C adapter -  USB A-USB A, etc.), then tried to change ports - USB C (CON), USB A 2.0, USB A 3.0, tried to change browsers and platform... It still doesn't work. Would you be so kind to help to find a reason? Any idea? Below data from my PiKVM.

**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.5
  - PiKVM platform PiKVM v3.3 HAT
  - Video capture type - CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
>kvmd 3.74-1
>kvmd-fan 0.20-1
>kvmd-oled 0.12-1
>kvmd-platform-v3-hdmi-rpi4 3.74-1
>kvmd-webterm 0.41-1
  - uStreamer version:`pacman -Q | grep ustreamer`
>ustreamer 5.3-1
  - Linux kernel: `uname -a`
>Linux pikvm.local 5.15.25-10-rpi-ARCH #1 SMP Wed Mar 16 06:04:18 MSK 2022 armv7l GNU/Linux

"pi-info.sh" from Community FAQ (https://docs.pikvm.org/community_faq/#troubleshooting):
`./pi-info.sh`
>15:37:57 up 11 min,  1 user,  load average: 0.25, 0.21, 0.14
>\# Raspberry Pi 4 Model B Rev 1.5
>
>CPU temp: 47.71'C
>GPU temp: 47.7'C
> 
>Throttled flags: 0x0
>
>Throttled now:  no
>Throttled past: no
>
>Undervoltage now:  no
>Undervoltage past: no
>
>Frequency capped now:  no
>Frequency capped past: no
>
>Version                 Package-Name
>\----------------------  -----------------------------
>20211222                firmware-raspberrypi-pikvm
>20220209.6342082        linux-firmware-pikvm
>20220209.6342082        linux-firmware-whence-pikvm
>20220301                raspberrypi-firmware-pikvm
>202203011               raspberrypi-bootloader-pikvm
>202203011               raspberrypi-bootloader-x-pikvm
>3.74                    kvmd
>0.20                    kvmd-fan
>0.12                    kvmd-oled
>3.74                    kvmd-platform-v3-hdmi-rpi4
>0.41                    kvmd-webterm
>5.3                     ustreamer
>1.21.6                  nginx-mainline
>2:2.10                  wpa_supplicant
>2022.02.18              wireless-regdb
>30.pre9                 wireless_tools

`cat /boot/config.txt`
>\# See /boot/overlays/README for all available options
>initramfs initramfs-linux.img followkernel
>
>hdmi_force_hotplug=1
>gpu_mem=128
>enable_uart=1
>dtoverlay=tc358743
>dtoverlay=disable-bt
>dtoverlay=dwc2,dr_mode=peripheral
>dtparam=act_led_gpio=13
> 
>\# HDMI audio capture
[... 265 lines cut ...]
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.661    stream] -- Using resolution: 1280x720
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.661    stream] -- Using format: UYVY
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.661    stream] -- Querying HW FPS changing is not supported
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.661    stream] -- Using IO method: MMAP
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.669    stream] -- Requested 5 device buffers, got 5
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- Capturing started
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- JPEG-0: Initializing encoder ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- JPEG-1: Initializing encoder ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- JPEG-2: Initializing encoder ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- Using JPEG quality: 80%
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- Creating pool JPEG with 3 workers ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.680    stream] -- Capturing ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.684    stream] -- H264: Configuring encoder: DMA=1 ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.684      jw-0] -- JPEG-0: Configuring encoder: DMA=0 ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.2.222/kvm/'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.71 Safari/53>
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.714      jw-1] -- JPEG-1: Configuring encoder: DMA=0 ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- => -- INFO  [92.727      jw-2] -- JPEG-2: Configuring encoder: DMA=0 ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='https://192.168.2.222/share/js/kvm/janus.js'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.>
>Apr 04 16:06:08 pikvm.local kvmd[394]: kvmd.apps.kvmd.streamer           INFO --- Got SIGUSR2, checking the stream state ...
>Apr 04 16:06:08 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [admin (token) / -] 'GET /auth/check HTTP/1.0' => 200; size=194 --- referer='-'; user_agent='Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/97.0.4692.71 Safari/537.36 Edg/97.0.1072.55'
>Apr 04 16:06:45 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [- / -] 'GET /auth/check HTTP/1.0' => 401; size=285 --- referer='-'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.3 Safari/605.1.15'
>Apr 04 16:07:27 pikvm.local kvmd[394]: kvmd.apps.kvmd.auth              ERROR --- Got access denied for user 'admin' from auth service 'htpasswd'
>Apr 04 16:07:27 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [- / 192.168.2.92] 'POST /auth/login HTTP/1.0' => 403; size=276 --- referer='https://192.168.2.222/login/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.3 Safari>
>Apr 04 16:08:09 pikvm.local kvmd[394]: kvmd.apps.kvmd.auth               INFO --- Authorized user 'admin' via auth service 'htpasswd'
>Apr 04 16:08:09 pikvm.local kvmd[394]: aiohttp.access                    INFO --- [- / 192.168.2.92] 'POST /auth/login HTTP/1.0' => 200; size=291 --- referer='https://192.168.2.222/login/'; user_agent='Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/15.3 Safari>

Maximka101, 2022-04-04:
Solved.
I found a problem. I changed some setting in BIOS (PS/2 keyboard and mouse section).
Motherboard is Asrock J4125-ITX.
Now all works!

closed 2022-04-04 by Maximka101

mdevaev, 2022-04-05:
:)

arch1mede, 2022-04-05:
> Solved. I found a problem. I changed some setting in BIOS (PS/2 keyboard and mouse section). Motherboard is Asrock J4125-ITX. Now all works!

What did you change in the BIOS?

arch1mede, 2022-04-05:
Thank you, this might be useful for others.