pikvm/pikvm #702: About remote USB mapping

weili71, 2022-04-07:
Is it possible to remotely map USB by installing programs on the remote OS

weili71, 2022-04-07:
Thus, the USB device on the remote computer is mapped to the local machine controlled by IP KVM

weili71, 2022-04-07:
> What do you mean?

Just like VMware, when USB is inserted, VMware will ask whether to map the USB device to the virtual machine, so that the USB device can be accessed in the virtual machine. Now the computer controlled by ipkvm is equivalent to virtual machine, and the remote computer is equivalent to physical machine. Now we hope to map the USB device on the remote computer to the computer controlled by ipkvm

weili71, 2022-04-07:
> Just like VMware, when USB is inserted, VMware will ask whether to map the USB device to the virtual machine, so that the USB device can be accessed in the virtual machine. Now the computer [...]

Can you see what I mean now？

mdevaev, 2022-04-08:
PiKVM USB emulation uses the regular OS driver so if you want to map into something special you can use the OS settings for this.

stevenruidigao, 2022-04-14:
I think what wilinz means is basically what I'm suggesting in #672.

mdevaev, 2022-04-30:
Yes, I think so.

closed 2022-04-30 by mdevaev