pikvm/pikvm #705: Cannot detect boot drive, when using  flash mode

techmadot, 2022-04-10:
**Describe the bug**

I'm using PiKVMHat. 
In my environment, I couldn't select the CD-ROM drive as the boot drive in the BIOS (UEFI) when selecting Flash mode. 
However, in CD-ROM drive mode, it was detected as a boot drive.

After Windows starts, Flash mode is working fine.

I want to use an iso, usb memory image larger than 2.2GB, so please give me some advice.

The image data was created from a physical USB stick (for the Ubuntu 21.04 installer). 
It has also been confirmed that it can be booted from this USB stick.
also, PiKVM Usb connects directly to the rear USB on the motherboard (not via Hub).

**To Reproduce**
Steps to reproduce the behavior, like:
1. Create an image from a bootable USB stick
2. Mount "Drive Mode" is Flash 
3. Reboot Target Computer , enter BIOS(UEFI) Screen
4. Check boot device selection menu, See error

**Expected behavior**

PiKVM devices are listed as boot devices

**Screenshots**

### Using Motherboard is ASRock B550 Phantom Gaming 4
[image]

This issue has occurred in other environments as well.

### Using Motherboard is A520M PRO-C DASH
[image]

when selected CD-ROM mode (Good case)
[image]

**Desktop (please complete the following information):**
 - OS: 
     - Windows10 
 - Browser [e.g. chrome, safari]
     - using Chrome  (100.0.4896.75
 - Version [e.g. 22]
 - VNC client (if used)
     - None 

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4]
      - RPi 4 (Memory 4GB)
  - PiKVM platform [e.g. v2-hdmi]
      - v3-hdmi-rpi4
  - Video capture type [e.g. CSI bridge]
      - CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
    - kvmd 3.80-1
    - kvmd-fan 0.20-1
    - kvmd-oled 0.12-1
    - kvmd-platform-v3-hdmi-rpi4 3.80-1
    - kvmd-webterm 0.41-1
  - uStreamer version: `pacman -Q | grep ustreamer`
    - ustreamer 5.3-1
  - Linux kernel: `uname -a`
    - Linux pikvm 5.15.25-15-rpi-ARCH #1 SMP Sat Apr 9 20:36:24 MSK 2022 armv7l GNU/Linux

**Additional context**
Add any other context about the problem here.

arch1mede, 2022-04-12:
I just tested this on my test NUC, I can see both cdrom and flash drives. However, I will point out that I needed to select legacy to view them.

mdevaev, 2022-04-12:
Did you try usb hard drive option

techmadot, 2022-04-13:
Thank you advices.
I tried it.

### case 1

USB Hard disk boot order has been selected as the highest priority.
But this failed.

### case 2 (Success)

I changed uefi mode to ''CSM".
And, boot mode selected Legacy + UEFI, then PiKVM CDROM appeared.

[image]

[image]

As a result, it doesn't seem to be recognized in UEFI Only mode.
Is this a specification? Is there any chance that it will be resolved in the future?

mdevaev, 2022-04-13:
It seems this is a compatibility problem. To fix it I need the motherboard with this bug to analyze the USB protocol features.

chuck-r, 2022-07-27:
Is the image you are trying to boot Secure Boot compatible? Many machines with Secure Boot enabled will not show boot options that cannot be verified via Secure Boot. Turning off Secure Boot may fix your issue.

mdevaev, 2022-08-01:
@techmadot please try the new KVMD 3.133+ and enable "writable" flag in MSD menu. It may fix this problem.

techmadot, 2022-08-05:
I tried last version.
but not fixed this problem.

[image]

if CSM is enabled, i can see bootable menu
[image]

mdevaev, 2022-08-05:
Got it. Tho this is not the same problem like with Steam Deck.

chuck-r, 2025-03-11:
Confirmed that it's working for me! I started work on a kernel patch back in 2016 or so for this, but found with the then-current version of f_mass_storage.c that I would have to write a DVD emulation layer and it never went anywhere (not to mention all of the locked-down specs and such). I'm glad to see this is finally working! This isn't a custom patch set and is an upstream kernel fix, correct?

mdevaev, 2025-03-12:
@chuck-r I wrote it myself based on few incomplete patches and my own understanding of SCSI.

mdevaev, 2025-03-12:
I suppose it can bt closed :)

closed 2025-03-12 by mdevaev