pikvm/pikvm #706: Annoying fan noise if kvmd-fan service is enabled

patrick7, 2022-04-10:
If the kvmd-fan service is enabled & started, there's an annoying noise from the fan. If disabled, the fan works too, but does not make any noise at all. Also the system is staying cool. 

**PiKVM info:**
  - Raspberry Pi 4 4GB
  - PiKVM HAT
  - KVMD version: 3.80.1
  - Linux kernel: 5.15.25-15-rpi-ARCH

mdevaev, 2022-04-10:
Try some tuning. Create file `/etc/conf.d/kvmd-fan` with this: `KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"`, and restart the service.

mdevaev, 2022-04-12:
Apparently fixed.

closed 2022-04-12 by mdevaev

kishkaru, 2022-09-03:
I could not find any user docs on how to configure the `kvmd-fan` service except this issue. There's no explanation on what `speed-idle`, `speed-low`, and `temp-low` are supposed to set.

Through trial & error, I was able to find the following worked best for the PWM fan in the steel case for pikvm:
```
KVMD_FAN_ARGS="--speed-idle=50 --speed-low=60 --temp-low=50"
```

Setting `speed-idle` below `50` causes major ball bearing noise on the fan. Setting it very low (`10`) causes massive PWM coil whine from the fan. Going higher than `50` caused too much fan noise. 50 worked the best.

Adjusting `speed-low` to higher than 60 also caused more noise.

Sandwraith0, 2022-09-13:
Hi there!

Long story short, I tried the above to adjust my fan speed (i.e. creating the file `/etc/conf.d/kvmd-fan` with `KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"`) but when I try to restart the kvmd-fan.service, I get the following error.

```
* kvmd-fan.service - PiKVM - A small fan controller daemon
     Loaded: loaded (/usr/lib/systemd/system/kvmd-fan.service; enabled; preset: disabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2022-09-13 23:55:27 MSK; 859ms ago
    Process: 2670 ExecStart=/usr/bin/kvmd-fan --config=?/etc/kvmd/fan.ini $KVMD_FAN_ARGS (code=exited, status=1/FAILURE)
   Main PID: 2670 (code=exited, status=1/FAILURE)
        CPU: 14ms
```
When I remove the `/etc/conf.d/kvmd-fan` file and restart the service, then the kvmd-fan.service starts just fine.

```
* kvmd-fan.service - PiKVM - A small fan controller daemon
     Loaded: loaded (/usr/lib/systemd/system/kvmd-fan.service; enabled; preset: disabled)
     Active: active (running) since Tue 2022-09-13 23:55:56 MSK; 2s ago
   Main PID: 2736 (kvmd-fan)
      Tasks: 2 (limit: 4020)
        CPU: 15ms
     CGroup: /system.slice/kvmd-fan.service
             `-2736 /usr/bin/kvmd-fan "--config=?/etc/kvmd/fan.ini"
```

What am I doing wrong here and were do I put `KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"` information?

Could someone help me out? Thanks in advance.

bigjohns97, 2023-01-14:
> KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"

This worked for me, thanks

Karbust, 2023-07-19:
> --speed-idle=10 --speed-low=10 --temp-low=50

Do you mind telling us where you put that? I'm using the BliKVM image and the fan is always blasting at full speed.

Already tried creating the file `/etc/kvmd/fan.ini` with `KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"` inside and setting `--speed-idle=10 --speed-low=10 --temp-low=50` directly on the service script, nothing worked...

mdevaev, 2023-07-19:
Bli* is a chinese fake that uses our trademarks without permission. We do not support this hardware. You should contact those who sold it to you.

Reiss-Cashmore, 2023-07-25:
Just for posterity there are some simple docs on the arguments in the kvmd-fan service

`      Hardware options:
	    --pwm-pin <N>  ─── GPIO pin for PWM. Default: %d.
	    --pwm-low <N>  ─── PWM low level. Default: %d.
	    --pwm-high <N>  ── PWM high level. Default: %d.
	    --pwm-soft <N>  ── Use software PWM with specified range 0...N. Default: disabled
	    --hall-pin <N>  ── GPIO pin for the Hall sensor. Default: disabled
	    --hall-bias <N>  ─ Hall pin bias: 0 = disabled, 1 = pull-down, 2 = pull-up. Default: %d.
	Fan control options:
	    --temp-hyst <T>  ───── Temperature hysteresis. Default: %.2f°C.
	    --temp-low <T>  ────── Lower temperature range limit. Default: %.2f°C.
	    --temp-high <T>  ───── Upper temperature range limit. Default: %.2f°C.
	    --speed-idle <N>  ──── Fan speed below of the range. Default: %.2f%%.
	    --speed-low <N>  ───── Lower fan speed range limit. Default: %.2f%%.
	    --speed-high <N>  ──── Upper fan speed range limit. Default: %.2f%%.
	    --speed-heat <N>  ──── Fan speed on overheating. Default: %.2f%%.
	    --speed-spin-up <N>  ─ Fan speed for spin-up. Default: %.2f%%.
	    --speed-const <N>  ─── Override the entire logic and set the constant speed. Default: disabled
	    -i|--interval <sec>  ─ Iterations delay. Default: %.2f.
	HTTP server options:
	    --unix <path> ─────── Path to UNIX socket for the /state request. Default: disabled
	    --unix-rm  ────────── Try to remove old UNIX socket file before binding. Default: disabled
	    --unix-mode <mode>  ─ Set UNIX socket file permissions (like 777). Default: disabled
	Config options:
	    -c|--config <path>  ─ Path to the INI config file. Default: disabled.
	Logging options:
	    --verbose  ─ Enable verbose messages. Default: disabled.
	    --debug  ─── Enable verbose and debug messages. Default: disabled.
	Help options:
	    -h|--help  ──── Print this text and exit.
	    -v|--version  ─ Print version and exit.`

plouie21, 2024-02-15:
That did the trick for me. Thank you very much. The commands for anyone interested

`
# cd /etc/conf.d/
# echo kvmd-fan
# nano kvmd-fan
(add line to file)KVMD_FAN_ARGS="--speed-idle=10 --speed-low=10 --temp-low=50"

# sudo systemctl restart kvmd-fan.service`

wesnm, 2024-10-21:
For the pikvm 4 Plus, the default settings resulted in horrible noise, like a strobe light by your desk, but with an idle and low speed set to 10 as above, the fan must have trouble spinning, because the logs immediately began filling up with errors:

```
Oct 21 15:46:32 pikvm kvmd-fan[7158]: -- INFO  [4598.186      loop] -- +++ Fan is spinning again +++
Oct 21 15:46:34 pikvm kvmd-fan[7158]: -- ERROR [4600.188      loop] -- !!! Fan is not spinning !!!
Oct 21 15:46:36 pikvm kvmd-fan[7158]: -- INFO  [4602.191      loop] -- +++ Fan is spinning again +++
Oct 21 15:46:38 pikvm kvmd-fan[7158]: -- ERROR [4604.193      loop] -- !!! Fan is not spinning !!!
Oct 21 15:46:40 pikvm kvmd-fan[7158]: -- INFO  [4606.194      loop] -- +++ Fan is spinning again +++
Oct 21 15:46:42 pikvm kvmd-fan[7158]: -- ERROR [4608.196      loop] -- !!! Fan is not spinning !!!

```

Setting all three speeds to 50 eliminated almost all of the clicks, and the fan itself is very quiet.