pikvm/pikvm #708: Q: Multiple subscribers, Auto Client Start?

doug62, 2022-04-17:
**Is your feature request related to a problem? Please describe.**
Q/Feature: This is a question and perhaps feature request

**Describe the solution you'd like**
I wish to use  this to broadcast my HDMI tv to multiple PIs (4 or so) around the house.  Want those devices to auto launch and auto reconnect to the host - headless.

Thanks

arch1mede, 2022-04-20:
Can you describe the use case? This cannot be turned into a chromecast device. This sounds more like a client side configuration.

doug62, 2022-04-20:
Use case is described above but: I wish to distribute a single HDMI feed to several screens through out the house - when one or more subscribing PIs turns on, it will automatically subscribe to the source and reconnect on failures.

arch1mede, 2022-04-20:
I am sorry but this still doesn't make any sense, can you provide explicit details on the use case. Please provide examples and or what you are attempting to emulate.

doug62, 2022-04-20:
Goal - Broadcast a single HDMI video/audio stream to multiple PIs that will connect to TVs around the house - an HDMI broadcast from a single source to many HDMI TVs.  Needs to be headless and automatic, start on boot and maintain the connection.

arch1mede, 2022-04-20:
All you managed to do was re-word whats already confusing therefor I am going to say that its not ever going to be possible to do what you want.

PiKVM is a 1:1 tool for troubleshooting. If you want end devices to log into the PiKVM then that's a client side configuration which is not supported. This is just my input, I will let the dev decide the final result and maybe he understands what you are asking for.

mdevaev, 2022-05-25:
I didn't understand anything, sorry.

closed 2022-06-26 by mdevaev