pikvm/pikvm #709: (hint) your fstab has been modified

mihcox, 2022-04-18:
**Describe the bug**
A clear and concise description of what the bug is.

When logging in via Putty to my PiKVM, and typing rw i see the below prompt:
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.

Another pikvm on my network does not have this issue

**To Reproduce**
Steps to reproduce the behavior, like:
1. Login to pikvm via ssh
2. Type rw
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

Expect to see nothing

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version [e.g. RPi 4] rpi 4b 2gb
  - PiKVM platform [e.g. v2-hdmi] v3.3 HAT
  - Video capture type [e.g. CSI bridge]
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**
Add any other context about the problem here.

arch1mede, 2022-04-20:
What did you change in your fstab file?

mihcox, 2022-04-20:
Nothing, this is a fresh install from the most current copy of pikvm
software installed via the pi sd card installer.

mdevaev, 2022-04-20:
I suppose you modified /etc/fstab. Run this:
```
rw
touch -m --date="2020-01-20" /etc/fstab
reboot
```
And everything will be fixed.

closed 2022-04-20 by mdevaev

A-s-a-d, 2022-06-18:
hi i'm installing archlinux i used mount /dev/sda1 /boot/efi and got the same  got same message tried your method it says bash: rw command not found
tried systemctl daemon-reload it says running in chroot, ignoring command 'daemon-reload. ho to solve it?

A-s-a-d, 2022-06-18:
i solved by going back with "exit" then "systemctl deamon-reload" then going back to chroot with "arch-chroot /mnt" and problem solved just redid couple of the previous commands to verifiy if everything was ok, everything seemed pk (no error)  then resumed where i left.

Kodlak15, 2022-06-22:
@A-s-a-d ran in to the same issue just now during Arch installation. Your solution worked for me as well. Thank you for sharing!

A-s-a-d, 2022-06-22:
I tried to install it through different methods but after installation when rebooting it always says "no boot media detected or install th e media ...." something like this. Someone said it could be because of bad partitions tried different methods of partition still same problem tried to install debian same problem tried kali linux which has graphical install and automatic partition still same problem. What could it be?

arch1mede, 2022-06-22:
Are you installing archlinux the distro or pikvm from our image repositories? I have burned the image many times and have never gotten these messages. What app are you using to burn the images?

A-s-a-d, 2022-06-22:
ARCHLINUX DISTRO
AND I'M USING RUFUS-3.18
I TRIED KALI LINUX ON 32 BIT LAPTOP WORKS FINE EXCEPT FOR GRAPHICS DRIVER