pikvm/pikvm #710: Boot visualisation issue with multiple video cards and displays

startergo, 2022-04-19:
**Describe the bug**
I have 2 video cards in the machine. One card is connected to a display and a 4K tv. The other card is connected to the HDMI input of the PiKVM. I can only see the boot screen from boot time if the monitor and the TV are disconnected. In Windows Pikvm and the display are mirrored, but apparently that setting is only valid for the operating system.

**To Reproduce**
Boot with 2 displays and Pikvm connected via HDMI.

**Expected behavior**
See boot screen with the option to go to BIOS in the Pikvm console.
**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Windows, Mac OS
 - Browser Edge, Safari
 - Version [e.g. 22]
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version RPi 4
  - PiKVM platform V3-hdmi-rpi4 3.82-1
  - Video capture type CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`  3.82-1
  - uStreamer version: `pacman -Q | grep ustreamer` 5.3-1
  - Linux kernel: `uname -a` Linux PiKVM 5.15.25-16-rpi-ARCH 

**Additional context**
Add any other context about the problem here.

arch1mede, 2022-04-20:
I am not sure you understand how this is supposed to work. PiKVM isn't made to be booted with 2 displays, only 1 as this is a troubleshooting tool so this works like adding an additional monitor. In most cases the main display will display the bios and the OS will either be mirrored or extended. It seems like you are mix and matching the behavior of the system and OS with PiKVM.

startergo, 2022-04-20:
> In most cases the main display will display the bios and the OS will either be mirrored

So if I have one display connected+pikvm should I be able to see the boot screen. As I said it only works on a headless system (only pikvm connected).

arch1mede, 2022-04-20:
Yes, that is the normal behavior of the system and as long as the PiKVM is connected to the main display output will you see the bios screen unless you somehow are able to change a setting and can display the bios on any and all gfx ports. I would hook a separate display to those other ports and review the system behavior, as long as you can get multiple screens/ports to display the bios then PiKVM will see that as well, as long as you are abiding by the correct resolution which is 1080p50hz or lower or 720p60hz.

startergo, 2022-04-20:
My [MOBO ](https://www.asus.com/us/Motherboards-Components/Motherboards/Workstation/WS-X299-PRO-SE/) supports ASMB9-iKVM, but of course It cannot redirect to the DGPU only to the serial port. Is there any way I can combine ASMB9-iKVM with piKVM in this respect?

startergo, 2022-04-20:
My ultimate goal is to be able to see the boot screen remotely without leaving it headless (unusable).

arch1mede, 2022-04-22:
I see, ok well its not possible and I am suspecting that it will never be possible but I will leave it up to the dev as he has much better  overview of this project than I do.

mdevaev, 2022-04-30:
Support for mirroring to different monitor outputs refers to the BIOS/UEFI functionality and there is nothing we can do here if the BIOS implements it incorrectly or does not implement it at all. PiKVM cannot force the process of displaying information on the screen in any way. It's physically impossible. I'm sorry, but all you can do is reduce the screen resolution in the BIOS, if it allows this.

closed 2022-04-30 by mdevaev