pikvm/pikvm #713: USB emulated device install windows can not find the driver

wilinz, 2022-04-19:
When installing windows 11 it shows:

A required media driver for your computer is missing. This could be a DVD, USB stick, or hard drive. If you have a CD, DVD or USB stick that contains the driver, insert it now.

Note: If the Windows installation media is in a DVD drive or a USB stick, you can safely remove it during this step.

Could not find any device drivers. Please make sure the installation media contains the correct drive and click OK.

[image]
[image]
"

NiKiZe, 2022-04-19:
Does it find your harddrive that you want to install to? You can use shift+F10 to get CMD and use diskpart to view partitions.

arch1mede, 2022-04-20:
Windows needs to be prepped and cannot be used natively, please review the MSD [document.](https://docs.pikvm.org/msd/#how-to-create-rw-flash-drive)

arch1mede, 2022-04-23:
Hi, did you get this working? I think it would be easier if you joined on [Discord](https://discord.gg/bpmXfz5)

mdevaev, 2022-04-30:
No answer.

closed 2022-04-30 by mdevaev