pikvm/pikvm #715: UEFI setup screen broken with ASRock H670 PG Riptide

syuu1228, 2022-04-21:
**Describe the bug**

I found how to fix this issue but reporting anyway,

Similar with #4 or #78, UEFI setup screen broken on ASRock H670 PG Riptide (everything else is working fine, only UEFI setup screen).

Not only right side of the screen doesn't appear correctly, the video output is completely frozen.

**To Reproduce**
1. Enter UEFI setup.

**Screenshots**
[image]
[image]

**PiKVM info:**
  - Raspberry Pi board version: RPi4
  - PiKVM platform: v3-hdmi
  - Video capture type: v3 HAT
  - KVMD version: kvmd 3.84-1 kvmd-fan 0.20-1 kvmd-oled 0.12-1 kvmd-platform-v3-hdmi-rpi4 3.84-1 kvmd-webterm 0.41-1
  - uStreamer version: ustreamer 5.3-1
  - Linux kernel: Linux pikvm 5.15.25-16-rpi-ARCH #1 SMP Mon Apr 11 04:59:20 MSK 2022 armv7l GNU/Linux

**Additional context**
GPU is MSI GeForce GTX 1050 2GT LP

**How to fix the issue**
Just like GA-H77-DS3H issue(#78) change EDID solved the issue.
After I applied ["1920x1080 as preferred. Useful for Gigabyte GA-H77-DS3H or Intel NUC"](https://docs.pikvm.org/edid/#default-edid) it works perfectly.

arch1mede, 2022-04-21:
Hi! Thank you for reporting this, can you change the title to add in the [SOLVED]? This way we can reference this to others. You can also remove the PiKVM info and additional context verbiage, as its non relevant.

mdevaev, 2022-04-21:
Edited.
@arch1mede could you document it on the edid page? Like Gigabyte motherboard

mdevaev, 2022-04-22:
Documented

closed 2022-04-22 by mdevaev

korhojoa, 2022-09-05:
> This also fix the same issue on MSI motherboard. After I applied [1920x1080 as preferred](https://docs.pikvm.org/edid/#default-edid) it works perfectly for me.
> 
> motherboard model: MSI MAG B660M MORTAR WIFI DDR4 video out: iGPU

Quite similar, MSI PRO Z690-A WIFI DDR4 with iGPU also was helped by using the 1920x1080 EDID