pikvm/pikvm #720: Add LDAP/AD user authentication

ohmer1, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
Using central authentication make managing users easier in enterprise env. It allow automatic centralized provisioning/deactivation and allow to enforce password policies.

**Describe the solution you'd like**
Support LDAP/AD user authentication. Configuration should permit to specify ldap filtering (like restricting access to users member of a specific user group).

**Describe alternatives you've considered**
I currently use local accounts, but this is a thing we try to avoid when possible.

mdevaev, 2022-04-26:
Sup. Right now we have two ways to make central authentication:
* Using `pam` auth plugin - you can configure PAM on PiKVM and use it to authenticate the users.
* HTTP auth: https://github.com/pikvm/kvmd-auth-server
The first way is required some configuration, the second one - some code from you. Will it fit or you need exactly LDAP or AD?

ohmer1, 2022-04-26:
I think yes. I could write an api function in our intranet that could be called by kvmd.

Thanks.

mdevaev, 2022-04-26:
:ok_hand:

closed 2022-04-26 by mdevaev

somethingstarted, 2024-02-07:
i would love this to be a standard feature, ldap / ad. that'd be the dream

mdevaev, 2024-02-07:
This is already implemented, but not documented: https://github.com/pikvm/kvmd/blob/master/kvmd/plugins/auth/ldap.py