pikvm/pikvm #722: Uploaded ISO file hash changes?!

WesBunton, 2022-04-27:
I want to automate the uploading of an ISO file for CD-ROM emulation. I've tried both the /api/msd/write API endpoint, and the non-WebUI method of SCP'ing a file to the .../images/ dir. In both cases, I can get the ISO file in place on the PiKVM, and can confirm that a hash calculation on the original file and the new remote file are the same after the upload. However, I then power cycle the unit that I have the PiKVM connected to. When it boots, the PiKVM stream starts up, and now the ISO hash is different, and I fail to complete the boot sequence due to a validation error.

I have the PiKVM v3 HAT from the kickstarter shipment. That is in use with the Pi 4. I have the v3-hdmi-rpi4-latest.img.xz image loaded on the PiKVM. The PiKVM is powered by a dedicated Pi power adapter to the USBC port on the HAT per the v3 setup guide. Not ATX device are setup/connected.

mdevaev, 2022-04-28:
I think this is a broken sd card. Try to replace it.

WesBunton, 2022-04-28:
Sorry - I should have mentioned that the mounted ISO hash calc does not change when I click through the Web UI to upload and connect the ISO. This only happens when I try to use the PiKVM automatically either with the API endpoint or SCP'ing the ISO file to the .../msd/images/ dir. This makes me think there's some really odd bug happening here...

mdevaev, 2022-04-28:
This can only happen if you interrupt the uploading. This bug has never been observed before and the uploader code has not changed for almost a year.

arch1mede, 2022-04-28:
I am also not able to replicate.

Please provide detailed step by step on exactly what you are doing. Including all commands and actions.

WesBunton, 2022-04-28:
Here's the exact steps that I perform:

1. Use the control MSD API endpoint to disconnect any previous image.
2. Use the get MSD state endpoint to generate a list of images.
3. Delete any existing images using the remote image endpoint.
4. Use 'mkisofs' to create a new ISO (this same code is working on other KVM's with virtual media, so I have faith in the ISO file).
5. Run 'md5sum' on the local ISO file.
6. Use the upload MSD (write, not write_remote) endpoint, to upload a new ISO file. This is not interrupted, I receive a 200 response and can see the new ISO in the web UI list of images.
7. Use the set MSD params endpoint to ensure the MSD is in CD-ROM mode, and select the new ISO.
8. Use the control MSD endpoint to connect the new ISO.
9. At this point I use another a Python SDK from Raritan, as the remote system that I'm controlling is powered by a Raritan PX3 PDU. The Python SDK let's me script the powering ON/OFF of specific outlets along the PDU. So here, I power ON the unit (as I the unit was in an OFF state prior to my PiKVM steps).
10. Once the unit comes up, it errors on boot because of a verification error of my boot media (ISO).
11. I can run 'md5sum /var/lib/kvmd/msd/images/*.iso', and see that the MD5 calculation is now different from the local ISO file I began with.

If I upload the ISO manually via the web UI, then this hash calc difference does not occur. I've also added in steps in my Python script to automatically run the md5sum remotely on the PiKVM over SSH, and I can confirm that when the ISO is initially uploaded, the MD5 calc matches, but once the stream starts after power ON, then the calc changes mysteriously (after the RO ISO is mounted).

mdevaev, 2022-04-29:
I'm confused. So, is the changing observed only on manual uploading via API?

WesBunton, 2022-04-29:
Update: Yes, I too was very confused. I seemed to be seeing differing behavior on the file integrity based on which methods I used to upload the ISO. I've even went so far as to automate the web UI upload via Selenium. I have more data today, and perhaps a suggestion of closing this.

I have two PiKVM's being used side-by-side. I was attempting an automation orchestration that utilized both, but I cut out the steps of the other one as soon as I began to struggle with this ISO upload on the first PiKVM. It wasn't until yesterday I simply ran my full automation steps, including the other PiKVM. To my surprise, the other one worked flawlessly, no ISO file integrity issues at all. And then during my work the troublesome PiKVM was rebooted. After the unit rebooted, I've since been running my ISO upload automation to both PiKVMs, and it's been working flawlessly on both.

Perhaps I should have tried the ol' reboot and retry much earlier during the slowly-losing-sanity process. I guess because the PiKVM seemed to be functioning totally normally otherwise, it hadn't occurred to me that perhaps a reboot would help. I'm also suspicious of the SD card. I'm prepared to swap out the SD card in the event that I see this behavior again. For now, all is well.