pikvm/pikvm #724: mjpeg vs. webRTC color range inconsistency

VoeGalore, 2022-05-04:
This is noticeable especially when looking at CLI with black background.
In mjpeg stream, black background is not completely black, color picker indicates RGB(16,16,16) a very faint grey.
In webrtc(H264) stream, black background is perfect black, RGB(0,0,0)

**Expected behavior**
mjpeg and webrtc should have same color range, same white/black levels.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser: Chrome
 - Version: 100

**PiKVM info:**
  - Raspberry Pi board version: Raspberry Pi 4 model B
  - PiKVM platform: v3.3 HAT
  - Video capture type: v3 HAT
  - KVMD version: 3.90-1
  - uStreamer version: 5.4-1
  - Linux kernel: 5.15.34-1-rpi-ARCH

**Additional context**
white level is also slightly different, in webrtc stream full white and full black is #ffffff and #000000, while in mjpeg stream they are slightly washed.

arch1mede, 2022-05-08:
My 2 cents on this, that's the nature of compression, this is a troubleshooting appliance first and foremost, I'm not sure I can see the value in maintaining the right color for troubleshooting. If you plan on using this outside its intended use case (bringing up a broken system) then I think you are using the incorrect project for your use case.

mdevaev, 2022-05-19:
Hints:
* https://en.wikipedia.org/wiki/YCbCr
* https://www.audioholics.com/home-theater-calibration/hdmi-black-levels-xvycc-rgb
* https://github.com/raspberrypi/libcamera-apps/commit/4bb16fee8c22b53326dd0eb3e571d0c5229ababb
* https://forums.raspberrypi.com/viewtopic.php?t=204694