pikvm/pikvm #725: Improve Alt+SysRq shortcuts

trapexit, 2022-05-05:
There are many Alt+SysRq commands and having shortcuts is very useful but some of the one's provided are either dangerous to have without confirmation or aren't as common as others.

'b', 'u', 'e', 'i' are all risky to have without confirmation

Ones that aren't present in the shortcuts but very useful are 'f' (oom killer), 'm' (dump memory), 'd' (locks), and 't'.

Naturally one can use the virtual keyboard but I find that those that I mentioned I use most often and it'd be nice to have shortcuts and have confirmation for them generally to keep from accidentally impacting the system.

mdevaev, 2022-05-14:
Okay

mdevaev, 2022-05-24:
Will be available in the next release

closed 2022-05-24 by mdevaev