pikvm/pikvm #726: Documentation for MSD emulation suggests storing images in /root when /var/lib/kvmd/msd might be more suitable

awkspace, 2022-05-06:
I'm using a prebuilt PiKVM v3, so maybe this is not applicable to all versions.

I'm following https://docs.pikvm.org/msd/ to create multiple mass storage devices, but I've noticed that the documentation suggests creating the additional flash drives under `/root`. On my PiKVM, `/` is rather small:

```
[root@pikvm ~]# df -h | grep mmcblk
/dev/mmcblk0p2  5.6G  2.0G  3.4G  37% /
/dev/mmcblk0p1  256M   42M  214M  17% /boot
/dev/mmcblk0p3   24G   32K   24G   1% /var/lib/kvmd/msd
```

It seems like `/var/lib/kvmd/msd` might be a more suitable path. Indeed, I have the feeling `msd` stands for Mass Storage Devices :)

I'd suggest changing the initial `rw` command (which corresponds to `/`) to `mount -o remount,rw /var/lib/kvmd/msd` and changing the other commands to reflect the creation of `flash.img` under `/var/lib/kvmd/msd`. Or at the very least, creating an `$MSD_DIR` variable that is used in subsequent commands, and suggesting it to be set to `/var/lib/kvmd/msd` on applicable versions.

arch1mede, 2022-05-08:
My 2 cents, this doc, along with the rest of the documents, gives one a starting point and that section gives an example on how to make a flash file used to make a flash drive, its required that you know enough about linux to know that if you need to make a flash file larger than your available space, that you need to use another location to allow the making of that file.

I don't feel we should accommodate the document change for your specific situation. If we added specific situations to the docs for every instance, it would be too long to read and no one would read them. (TL;DR)

mdevaev, 2022-05-14:
Everything that @arch1mede wrote, plus /var/lib/msd can't store RW images. So we will not change this.

closed 2022-05-14 by mdevaev