pikvm/pikvm #730: Video stream shows pink and white snow.

bobtiji, 2022-05-09:
**Describe the bug**
Very often the video stream will completely be covered by white and pink snow

**To Reproduce**
Steps to reproduce the behavior, like:
Mostly happens when i reset my server via pikvm v3.
I type sudo reboot in my cli,
I see the reboot process ( at this point the stream works)
a few seconds after reboot the screen goes blank and fills with white pink snow

**Expected behavior**
The display is supposed to show me the bios prompt at that point

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Desktop (please complete the following information):**
 - OS: pop os 22.04
 - Browser brave1.37.116
 - VNC client (if used)

**PiKVM info:**
  - Raspberry Pi board version pi 4 4gb
  - PiKVM platform  v3.3
  - Video capture type: dont know the stock one
  - KVMD version: 3.90-1
  - uStreamer version: 5.4-1
  - Linux kernel: Linux pikvm 5.15.34-1-rpi-ARCH #1

**Additional context**
have tried with different hdmi cables
have tried with the igpu and the 1070ti
have tried resetting the video cable in the pi board and the hat

mdevaev, 2022-05-11:
Do you use VGA adapter or something like that?

bobtiji, 2022-05-11:
no hdmi cable

mdevaev, 2022-05-13:
Please check two ways to workaround this:
* Reboot PiKVM - will the white snow keep after this.
* Run these commands and check the same:
```
/usr/bin/v4l2-ctl --device=/dev/kvmd-video --clear-edid
(wait 10 seconds)
/usr/bin/v4l2-ctl --device=/dev/kvmd-video --set-edid=file=/etc/kvmd/tc358743-edid.hex --fix-edid-checksums --info-edid
```

bobtiji, 2022-05-13:
Thanks, I'll check it out.

mdevaev, 2022-05-15:
Sup. Any news?

bobtiji, 2022-05-15:
am going to test it right now

bobtiji, 2022-05-15:
Still give this
but framerate is unstuck for a few seconds after the commands

mdevaev, 2022-05-19:
Do I understand correctly that even rebooting PiKVM doesn't help?

Try using another HDMI source, such as laptop. What happens if you reboot that another source? If the same artifacts appear, then perhaps it is a PiKVM defect and we will replace it.

bobtiji, 2022-05-19:
alright I'll give it a try

bobtiji, 2022-05-22:
It didn't help back then.
Now everything seems fine. I rebooted the server with the webui opened and
no problem with the screen anymore.
Your clear edid workaround and pacman update seem to have worked.

Thanks for the good service and for a great product.

mdevaev, 2022-05-22:
Glad to hear it!

closed 2022-05-24 by mdevaev