pikvm/pikvm #732: The ability to switch between Mouse modes through the UI

youcangetme, 2022-05-12:
**Is your feature request related to a problem? Please describe.**
I will use my pikvm with both a PC and a mobile device. Without the ability to switch between the absolute and relative mouse modes I can't use the Web UI from a mobile device because of the touch screen incompatibility. My laptop also has two touch screens and a pen, none of which work as well.

**Describe the solution you'd like**
Under "Mouse polling rate" add a switch to select either absolute or relative mouse modes

**Describe alternatives you've considered**
Just optimize for the PC which I use more often and pray the client doesn't call when I am at lunch!

**Additional context**
So far the pikvm has been everything I need and just the ability to switch between mouse modes being the only bummer.

youcangetme, 2022-05-12:
> It's already there. Please use the dual mouse mode:
> https://docs.pikvm.org/mouse/#relative-mouse-on-v2-platform-otg-hid

But I have to go in and use the term to enable the mouse when I am on
mobile. Then I have access to the relative position.

I have to switch it back when I log in with a PC by going back into the
term, set rw and make the change.

So, yes. It is there but it isn't switchable from the UI unless you are
referencing something that isn't in the documentation.

mdevaev, 2022-05-12:
Please read it carefully. You need to enable dual mode ONCE in config, after that there will be available a switch.

youcangetme, 2022-05-12:
Ok, I see it now. The whole "dual mode" part was confusing.

Maybe "Enable Mouse Mode Button" would help folks realize the intent.

In big bold letters with arrows pointing at it, lol!

Thank you for the help.

mdevaev, 2022-05-13:
:ok_hand: