pikvm/pikvm #734: HDMI video appears to have sync issue

alexg0240, 2022-05-14:
When I connect the PiKVM to a PC, the video appears normal.  But when I connect it to a video stream, it appears to be out of sync.  Does anyone know how to resolve this? 

I'm taking an inhouse video from a sdi source and running it thru a sdi to hdmi blackmagic converter and then going into the hdmi on the PiKVM.  I see the video coming thru but it looks like its out of sync maybe.  there shouldn't be any encryption at this point... basically just taking it off a camera.

**Expected behavior**
I was expecting it to simply pass the video thru... I ran the same video into a monitor and video capture bm device into a pc using obs... don't know why its not working.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Software: OBS
 - Black Magic sdi to usb c capture device

Black Magic Design sdi to hdmi into the PiKVM

**PiKVM info:**
  - Raspberry Pi board version RP4
  - PiKVM platform v3
  - Video capture type CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
  - uStreamer version: `pacman -Q | grep ustreamer`
  - Linux kernel: `uname -a`

**Additional context**

mdevaev, 2022-05-14:
Try different edids: https://docs.pikvm.org/edid/

I think this is a glitch of the converter.

mdevaev, 2022-05-19:
Sup?

mdevaev, 2022-05-22:
No answer, solved I guess.

closed 2022-05-22 by mdevaev