pikvm/pikvm #737: Option for permanent OCR

mhameed, 2022-05-20:
Hi, and thanks for pikvm, a great project!
I have just received my pikvmv3, now assembled thanks to a friend.

I am blind and use a screenreader, I have read that pikvm can use OCR when selecting regions of the screen, to enable copy/paste.
I was wondering if/how I can make this option permanent, so that I can read the exposed text using my screenreader.

Many thanks

mdevaev, 2022-05-22:
Sup. Sorry, but I don't think it's possible. OCR is a haevy operation and takes very much time for one image, like 2-3 seconds.

mdevaev, 2022-05-24:
If you have any suggestions, I will be glad to hear it here. Until that, I'm closing this issue.

mhameed, 2022-05-26:
Thank you for your reply.

Can we please add a shortcut which can be pressed which will send the entire screen to OCR?

I know it is not efficient, or clean, but it is better to press and wait a few seconds, than not 
having any access at all, or waiting for sighted assistance.

This would be very useful for blind people when there is something wrong with their computers, or 
they need to change something in the bios.

Thanks for your time.

mdevaev, 2022-05-26:
I think yes.

123swk123, 2022-06-02:
just a thought if we can define region of interest(ROI) it will reduce the work load on CPU for OCR processing 

of course it depends on the ROI size

connected201, 2022-07-24:
maybe can connect to https://ocr.space/OCRAPI and send images to recognize only area with needed data and return results from server :)

mdevaev, 2022-07-24:
Using cloud services for such things is a bad idea in terms of security and privacy.

wnark, 2022-08-31:
Because the ocr api can be built with docker, we can try to see if we can use the local ocr api server, so that the security problem can be solved.