pikvm/pikvm #738: kvmd-fan takes over *both* hardware PWM channels

jacopo-j, 2022-05-21:
**Describe the bug**
`kvmd-fan`, which uses the `wiringpi` library, takes over both hardware PWM channels regardless of the pin where the fan is connected. This results in devices (such as a servo) connected to the other channel to stop working. As discussed on Discord, the program may need to switch to `bcm2835`.

**To Reproduce**
Steps to reproduce the behavior, like:
1. Setup a servo on PWM channel 0
2. Connect a fan to PWM channel 1
3. Start `kvmd-fan` passing the correct fan pin as argument
4. The servo stops working

**Expected behavior**
Starting `kvmd-fan` does not affect the other PWM channel

**PiKVM info:**
  - Raspberry Pi board version: RPi 4
  - PiKVM platform: v2-hdmi
  - Video capture type: CSI bridge
  - KVMD version: `3.61-1`
  - uStreamer version: `5.1-2`
  - Linux kernel: `5.15.25-10-rpi-ARCH`

mdevaev, 2022-06-10:
It can't be fixed since both pwm on pi uses the same clock source: https://www.airspayce.com/mikem/bcm2835/
> **Both PWM channels are driven by the same PWM clock**, whose clock dvider can be varied using [bcm2835_pwm_set_clock()](https://www.airspayce.com/mikem/bcm2835/group__pwm.html#ga4487f4e26e57ea3697a57cf52b8de35b). Each channel can be separately enabled with [bcm2835_pwm_set_mode()](https://www.airspayce.com/mikem/bcm2835/group__pwm.html#ga7e2eddac472b6d81e0ba7dbf165672cb). The average output of the PWM channel is determined by the ratio of DATA/RANGE for that channel. Use [bcm2835_pwm_set_range()](https://www.airspayce.com/mikem/bcm2835/group__pwm.html#ga9f5ca04d4f859d96f1205a03c79de7ce) to set the range and [bcm2835_pwm_set_data()](https://www.airspayce.com/mikem/bcm2835/group__pwm.html#ga4f445a60ace471a01d0cc4015bcea1dd) to set the data in that ratio

I believe that the easiest way would be to fix the servo plugin to optionally use software pwm.

NyaMisty, 2022-11-30:
Hi, I believe instead of make servo plugin to use software pwm, why don't we make to fan use software pwm?

mdevaev, 2022-12-02:
Since 0.24 kvmd-fan can work in soft pwm mode. To enable it add these lines to `/etc/kvmd/fan.ini`:
```ini
[main]
pwm_soft = 80
```

closed 2022-12-02 by mdevaev