pikvm/pikvm #741: Use webcam and microphone from the guest machine inside the host machine

ale-novo, 2022-05-26:
Hi, just found out about PIKVM.
Im thinking to use it to remotely connect to my work laptop. I have periodically conference calls. 
If the work laptop that is connected to PIKVM is the host machine, and my remote laptop im connecting from is the guest. 
Can i use my guests webcam and microphone to take conference calls from my guest machine just as if i was sitting in front of my host machine?
The same way keyboard and mouse work? cheers

arch1mede, 2022-05-26:
Asked and answered on Discord, sound is not yet implemented and still WIP, you can of course use a webcam connected to the pikvm to point to a screen BUT this project does not broadcast streams to a target. I do not believe there is any such thing that can do this.

arch1mede, 2022-06-09:
Just an FYI, sound will be available for all officially within days/weeks but there will not be mic support, someone suggested using jackaudio.org but you will be on your own for this setup. You can view what a community member made mention of it [here](https://discord.com/channels/580094191938437144/880066200959328328/983262344023015474)

mdevaev, 2022-06-28:
This about a generic USB passthrough: #672

closed 2022-06-28 by mdevaev

mdevaev, 2025-01-20:
Official [PiKVM V3 and V4 devices](https://pikvm.org/buy) gets two-way audio with microphone.

Now you can use voice applications remotely, or you can just emotionally tell the server what you think about it.

**https://docs.pikvm.org/audio**

*PS: I know that many people wanted this feature. I spent all the New Year holidays on it, so if you found PiKVM useful and finally got what you wanted, you can support my work on PiKVM or just give me a beer with donation to [Patreon](https://www.patreon.com/pikvm) or [Paypal](https://paypal.me/pikvm). Happy PiKVMing!*

Majestic7979, 2025-02-04:
> Official [PiKVM V3 and V4 devices](https://pikvm.org/buy) gets two-way audio with microphone.
>
> Now you can use voice applications remotely, or you can just emotionally tell the server what you think [...]

You deserve it. I really like you as a person, you are dedicated. You helped me in the past with an issue. Your project really helped me a lot.

[image]

Enjoy the beer! I don't know how much beer is in your country but I hope this is enough for a big glass!!