pikvm/pikvm #742: oled service failed to start

123swk123, 2022-05-31:
**Describe the bug**
kvmd-oled.service failed to start after system upgrade `pacman -syu`

**To Reproduce**
Steps to reproduce the behavior, like:
1. perform system upgrade `pacman -syu`
2. After successful upgrade reboot `reboot now`
4. you should See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**error log**
```sh
May 31 19:08:57 pikvm-3.local kvmd-oled[892]: Traceback (most recent call last):
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:   File "/usr/bin/kvmd-oled", line 214, in <module>
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:     main()
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:   File "/usr/bin/kvmd-oled", line 159, in main
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:     device = luma_cmdline.create_device(options)
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:   File "/usr/lib/python3.10/site-packages/luma/core/cmdline.py", line 238, in create_device
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:     device = Device(device=args.framebuffer_device, **params)
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 208, in __init__
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:     self.id = self.__get_display_id(device)
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 246, in __get_display_id
May 31 19:08:57 pikvm-3.local kvmd-oled[892]:     raise luma.core.error.DeviceNotFoundError(
May 31 19:08:57 pikvm-3.local kvmd-oled[892]: luma.core.error.DeviceNotFoundError: Invalid/unsupported framebuffer: /dev/fd0
May 31 19:08:58 pikvm-3.local systemd[1]: kvmd-oled.service: Control process exited, code=exited, status=1/FAILURE
May 31 19:08:58 pikvm-3.local systemd[1]: kvmd-oled.service: Failed with result 'exit-code'.
May 31 19:08:58 pikvm-3.local systemd[1]: Failed to start Pi-KVM - Small OLED daemon.
```
**PiKVM info:**
  - Raspberry Pi 4 Model B Rev 1.5
  - PiKVM platform: v3-hdmi-rpi4
  - Video capture type: v3 HAT
  - KVMD version: 
    - kvmd 3.93-1
    - kvmd-fan 0.20-1
    - kvmd-oled 0.12-1
    - kvmd-platform-v3-hdmi-rpi4 3.93-1
    - kvmd-webterm 0.42-1
  - uStreamer version: ustreamer 5.4-1
  - Linux kernel: 5.15.34-1-rpi-ARCH # 1 SMP Fri Apr 22 21:35:22 MSK 2022 armv7l GNU/Linux

arch1mede, 2022-05-31:
I suggest you download the box image and reflash it using this image and or I would ask that you come onto Discord to discuss, unable to repo this issue.

Box image can be found here:
https://files.pikvm.org/images/

Discord link:
https://discord.gg/bpmXfz5

123swk123, 2022-06-01:
- installed box latest image on a 32GB SD card
- ran system upgrade `pacman -Syuy` [system-upgrade.typescript.txt](https://github.com/pikvm/pikvm/files/8812370/system-upgrade.typescript.txt)
- after reboot end up in same issue [kvmd-oled.typescript.txt](https://github.com/pikvm/pikvm/files/8812371/kvmd-oled.typescript.txt)

enclosed typescript file captured using `script` might help you for replay

mdevaev, 2022-06-01:
This is upstream bug, I'll make a workaround: https://bugs.archlinux.org/task/74933

json81, 2022-06-10:
Hi!
I think I have the same problem and `pacman -Syu` did not solve it.

```
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 pikvm is up to date
:: Starting full system upgrade...
 there is nothing to do
```

```
systemctl enable --now kvmd-oled kvmd-oled-reboot kvmd-oled-shutdown
Job for kvmd-oled.service failed because the control process exited with error code.
See "systemctl status kvmd-oled.service" and "journalctl -xeu kvmd-oled.service" for details.
```

```
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     main()
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/bin/kvmd-oled", line 159, in main
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     device = luma_cmdline.create_device(options)
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/lib/python3.10/site-packages/luma/core/cmdline.py", line 246, in create_device
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     device = Device(serial_interface=interface(), **params)
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/lib/python3.10/site-packages/luma/oled/device/__init__.py", line 176, in __init__
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     self.command(
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 48, in command
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     self._serial_interface.command(*cmd)
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/lib/python3.10/site-packages/luma/core/interface/serial.py", line 96, in command
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     self._bus.write_i2c_block_data(self._addr, self._cmd_mode,
Jun 10 22:52:24 pikvm kvmd-oled[4336]:   File "/usr/lib/python3.10/site-packages/smbus2/smbus2.py", line 643, in write_i2c_block_data
Jun 10 22:52:24 pikvm kvmd-oled[4336]:     ioctl(self.fd, I2C_SMBUS, msg)
Jun 10 22:52:24 pikvm kvmd-oled[4336]: TimeoutError: [Errno 110] Connection timed out
Jun 10 22:52:25 pikvm systemd[1]: kvmd-oled.service: Control process exited, code=exited, status=1/FAILURE
Jun 10 22:52:25 pikvm systemd[1]: kvmd-oled.service: Failed with result 'exit-code'.
Jun 10 22:52:25 pikvm systemd[1]: Failed to start Pi-KVM - Small OLED daemon.
Jun 10 22:52:28 pikvm systemd[1]: kvmd-oled.service: Scheduled restart job, restart counter is at 381.
Jun 10 22:52:28 pikvm systemd[1]: Stopped Pi-KVM - Small OLED daemon.
Jun 10 22:52:28 pikvm systemd[1]: Starting Pi-KVM - Small OLED daemon...
Jun 10 22:52:30 pikvm kvmd-oled[4348]: Traceback (most recent call last):
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/bin/kvmd-oled", line 214, in <module>
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     main()
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/bin/kvmd-oled", line 159, in main
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     device = luma_cmdline.create_device(options)
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/lib/python3.10/site-packages/luma/core/cmdline.py", line 246, in create_device
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     device = Device(serial_interface=interface(), **params)
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/lib/python3.10/site-packages/luma/oled/device/__init__.py", line 176, in __init__
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     self.command(
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/lib/python3.10/site-packages/luma/core/device.py", line 48, in command
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     self._serial_interface.command(*cmd)
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/lib/python3.10/site-packages/luma/core/interface/serial.py", line 96, in command
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     self._bus.write_i2c_block_data(self._addr, self._cmd_mode,
Jun 10 22:52:30 pikvm kvmd-oled[4348]:   File "/usr/lib/python3.10/site-packages/smbus2/smbus2.py", line 643, in write_i2c_block_data
Jun 10 22:52:30 pikvm kvmd-oled[4348]:     ioctl(self.fd, I2C_SMBUS, msg)
Jun 10 22:52:30 pikvm kvmd-oled[4348]: TimeoutError: [Errno 110] Connection timed out
Jun 10 22:52:31 pikvm systemd[1]: kvmd-oled.service: Control process exited, code=exited, status=1/FAILURE
Jun 10 22:52:31 pikvm systemd[1]: kvmd-oled.service: Failed with result 'exit-code'.
Jun 10 22:52:31 pikvm systemd[1]: Failed to start Pi-KVM - Small OLED daemon.
```

Do I need to change the update source?

Best Regards
Anders Jansson
Best regards

arch1mede, 2022-06-10:
I would recommend that you relfash and update it again. That would be the fastest fix.

json81, 2022-06-11:
I found the problem and it is now working. It was not connected correctly.

You should add the OLED Pins to the documentation:

[image]

arch1mede, 2022-06-11:
I am confused now, how did you have this hooked up wrong if you have the HAT? Its rather apparent what this is for when you reference the video on how to assemble it.

json81, 2022-06-11:
Well, the kit I bought didn't come with a OLED so I bought one separate. This OLED had cables. I didn't realize that I could use the pass thru pins instead of the PiKVM L2C display connector.