pikvm/pikvm #749: RTC errors in system log

taintedkernel, 2022-06-16:
**Describe the bug**
Built a new PiKVM on a RPi 4 and after bootup, I was trying to figure out why my OLED display was not working and I see repeating entries in the system log:
`
rtc-pcf8563 1-0051: low voltage detected, date/time is not reliable.
`

**To Reproduce**
Steps to reproduce the behavior, like:
1. Boot system
2. Run `dmesg`

**Expected behavior**
Error messages not in system log

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - n/a

**PiKVM info:**
  - RPi 4
  - PiKVM v3.3 HAT
  - Video capture type: CSI Bridge
  - KVMD version: kvmd 3.61-1
  - uStreamer version: ustreamer 5.1-2
  - Linux pikvm 5.15.25-10-rpi-ARCH #1 SMP Wed Mar 16 06:04:18 MSK 2022 armv7l GNU/Linux

mdevaev, 2022-06-16:
Hello. The supercapacitor is gaining charge and the problem will disappear by itself.

arch1mede, 2022-06-16:
Please see this [document](https://docs.pikvm.org/faq/#hardware-problems-wi-fi-atx-etc).