pikvm/pikvm #756: Brute force protection on the web interface

SharkWipf, 2022-06-25:
**Is your feature request related to a problem? Please describe.**
Currently, there's no form of brute force protection on the web interface, which makes it highly inadvisable to expose to the internet.

**Describe the solution you'd like**
Brute force protection on the web interface.

**Describe alternatives you've considered**
It is of course never recommended to directly expose a BMC-like device to the open internet, as this type of device has a long and repeated history of having critical bugs that allow full system compromise, but not having any form of brute force protection at all is just unnecessarily tempting fate.

**Additional context**
Just finally got around to setting up one of my PiKVMs and it's really nice all around, just a few minor issues like this that stand out.

arch1mede, 2022-06-26:
My 2/5 cents on this, OS hardening shouldn't be up to the dev but rather to the end user.

mdevaev, 2022-06-26:
We have to provide the very opportunity to do this, now it is not there.

mdevaev, 2023-06-07:
Right now we have 2FA. Would that suit you as a defense?

https://docs.pikvm.org/auth/#two-factor-authentication

SharkWipf, 2023-06-08:
Hmm, honestly, the ability to have 2FA is great, but it doesn't do much for the out of the box experience.  
I am not exposing my PiKVM directly in the first place, and if I did I'd definitely set up 2fa at the very least.

However, this is more about people less aware of the risks involved, who'd just buy a PiKVM, hook it up, forward it and call it a day without properly reading the docs and/or warnings.  
One could argue they might get what's coming to them, but it certainly wouldn't hurt improving the out of the box security.  
It may be stupid to directly expose your PiKVM to the internet without precautions, but that doesn't mean it should be an easy target for hackers when people do it anyway.

I'd expect it shouldn't take much, if any, custom development, I'm pretty sure one could just slap on fail2ban and give it a rule to monitor the webserver logs and hand out temp IP bans to offenders.