pikvm/pikvm #758: Error: Unknown development platform 'atmelavr' while flashing Arduino HID

awolf666, 2022-06-27:
I used a PI 2 to setup PIKVM v0. Did everything like in the manual described. There were a few warnings and errors while building the image but the pi booted with the created image. When I tried to flash the Arduino (Pro Micro Atmel Mega 32U4) the following error occured:

[root@pikvm2 hid]# make spi
make _build E=spi
make[1]: Entering directory '/root/hid'
rm -f .current
platformio run --environment spi
************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
************************************************************************************************************************

Processing spi (platform: atmelavr; board: micro; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing atmelavr
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: atmelavr @ 4.0.0 has been installed!
Error: Unknown development platform 'atmelavr'
make[1]: *** [Makefile:9: _build] Error 1
make[1]: Leaving directory '/root/hid'
make: *** [Makefile:4: spi] Error 2

I removed "hid" and ".platformio" folders, reinstalled platformio via script from official homepage.
Still the same error

**PiKVM info:**
  - RPi 2
  - v0-hdmi
  - CSI bridge
  - KVMD version: `pacman -Q | grep kvmd`
kvmd 3.117-1
kvmd-fan 0.22-1
kvmd-oled 0.12-1
kvmd-platform-v0-hdmi-rpi2 3.117-1
kvmd-webterm 0.42-1
  - uStreamer version: ustreamer 5.9-1
  - Linux kernel: Linux pikvm2 5.15.34-2-rpi-ARCH #1 SMP Fri Jun 3 06:14:48 MSK 2022 armv7l GNU/Linux

mdevaev, 2022-06-27:
I've build the new platformio. Update OS, remove old `hid` directory and retry. It will work.

closed 2022-06-27 by mdevaev

awolf666, 2022-06-28:
Please assume that I'm a total noob. (because I am)

I rebuild the image following these instructions:
https://docs.pikvm.org/building_os/

--> QUESTION: I just deleted the old "os" folder.  Are there any other remnants (e.g. docker container?) hiding elsewhere? Because flashing the firnware didn't work initially. Still got the old error.

Next flashing the Arduino:
https://docs.pikvm.org/flashing_hid/
Still got the same error.
After running "pacman -Syu" it worked. Perhaps that command should be integrated into the instructions above. (I never had anything done with arch or pacman (because I am noob) so I didn't know how to run a system upgrade)

Regards

mdevaev, 2022-06-28:
This is a temporary phenomenon, it won't be needed anymore.