pikvm/pikvm #759: What parameters need to be enabled for a self built stun server?

TelDragon, 2022-06-29:
Kvmd Janus detection fails when I start the server with the stun default parameter
It is OK to execute the command manually.
`/usr/bin/janus --disable-colors --plugins-folder=/usr/lib/ustreamer/janus --configs-folder=/etc/kvmd/janus --interface=192.168.1.17 --stun-server=xx.xxx.xxx.xx:3478`

kvmd/apps/janus/stun.py
` async def get_info(self, src_ip: str, src_port: int) -> Tuple[str, str]:

        (family, _, _, _, addr) = socket.getaddrinfo(src_ip, src_port, type=socket.SOCK_DGRAM)[0]
        try:
            with socket.socket(family, socket.SOCK_DGRAM) as self.__sock:
                self.__sock.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
                self.__sock.settimeout(self.__timeout)
                self.__sock.bind(addr)
                (nat_type, response) = await self.__get_nat_type(src_ip)
                return (nat_type, (response.ext.ip if response.ext is not None else ""))
        finally:
            self.__sock = None`

arch1mede, 2022-06-29:
Hi, are you aware of the [static-janus](https://docs.pikvm.org/webrtc/#custom-janus-config) option?

TelDragon, 2022-06-30:
```
systemctl disable kvmd-janus
systemctl enable kvmd-janus-static
```

/etc/kvmd/janus/janus.jcfg

```
general: {
        debug_level = 4
}
nat: {
        nice_debug = false
        ignore_mdns = true
+ stun_server = xxx.xxx.xx.xx
+ stun_prot = 3478        
}
media: {
        ipv6 = false
        ipv6_linklocal = false
        min_nack_queue = 2000
        rtp_port_range = "20000-40000"
        no_media_timer = 0
        slowlink_threshold = 10
        twcc_period = 100
        dtls_timeout = 1000
        nack_optimizations = true
}

systemctl start kvmd-janus-static
```

Is that so?

TelDragon, 2022-06-30:
excellent! Thank you ~